ensenso/ros_driver #1: Relaiton to a community version of ROS driver package

130s, 2017-11-11:
Thanks for making this package opensource!

As there has been a [ROS driver pkg that was made and has been maintained by community](https://github.com/crigroup/ensenso), I assume a lot of ensenso-ROS users are relying on it. Are these related, API inter-operable, or planned to be so?

Thanks!

CC @fsuarez6

fsuarez6, 2017-11-12:
This metapackage ([ensenso/ros_driver](https://github.com/ensenso/ros_driver)) is not related to the package [crigroup/ensenso](https://github.com/crigroup/ensenso). 
As a matter of fact, their design is quite different. [ensenso/ros_driver](https://github.com/ensenso/ros_driver) uses ROS actions for everything and [crigroup/ensenso](https://github.com/crigroup/ensenso) uses ROS services and dynamic reconfigure.

We could select the 'best' pieces and merge them into one unified (meta)package.

saierd, 2017-11-13:
The two packages are not related and we currently don't plan to make them compatible.

This package was designed in collaboration with two companies that currently use their own ROS packages for Ensenso cameras. We also looked into the existing open source packages and the functionality they provide. The result hopefully fits the needs of many users.

Of course, if you want to have additional functionality or want to adopt some features from the crigroup package, we are open for discussion and contributions.

130s, 2017-11-15:
Thank you two for clarification.

As a member of opensource community I have no more words than appreciation for making your work public. I have no right to push you to make extra effort for consolidating packages, but if that could ever happen then I can suggest to my team to test it out.

Just a background of my opening this ticket; Almost 4 years ago I found 3 separate packages of UEYE driver (again IDS product, just a coincidence) and [raise an attention](http://lists.ros.org/pipermail/ros-users/2013-December/068075.html). Developers of at least two of the packages were willing to collaborate in the beginning, which sadly but understandably didn't work out probably for a number of unavoidable reasons. If things go better this time, that will be great.

To the least, I opened https://github.com/ros/rosdistro/pull/16373 as suggested in the linked thread, to make `ensenso` package more visible to the community. Also, maybe I suggest in both packages document, we as a user want to see reference to each package (as `ueye_cam` package does [here](http://wiki.ros.org/ueye_cam#Footnote:_Relationship_to_other_ROS_wrappers_for_UEye_cameras)) to avoid future user confusion.

gavanderhoorn, 2017-11-15:
It's a bit unfortunate the package is called `ensenso`. That doesn't really follow [REP-144](http://www.ros.org/reps/rep-0144.html).

fsuarez6, 2017-11-16:
> It's a bit unfortunate the package is called `ensenso`. That doesn't really follow [REP-144](http://www.ros.org/reps/rep-0144.html).

Agree. That's why we are planning to convert it into a metapackage with the following structure:
```
ensenso
├── ensenso_driver
└── ensenso_msgs
```

gavanderhoorn, 2017-11-16:
That would seem to be even more unfortunate, as it would then clash with the pkgs in this repository?

closed 2021-03-12 by saierd