ensenso/ros_driver #2: Consider not using 'map' as the default frame

gavanderhoorn, 2017-11-12:
`map` is really more typically used with mobile robots to provide a reference for a global frame that associated with a map. See also [REP-105: Coordinate Frames for Mobile Platforms - map](http://www.ros.org/reps/rep-0105.html#map).

It *is* the default frame that RViz starts with, but this is typically worked around by either providing a custom RViz config file that already uses a different frame, or by documenting the steps needed to change the *Fixed Frame*. Another option is to use the `-f` or `--fixed-frame` command line option, which should allow you to specificy a frame while starting RViz.

To be more in-line with other ROS camera drivers, perhaps `${prefix}_rgb_optical_frame` and `${prefix}_depth_optical_frame` (as needed) might be more appropriate.

saierd, 2017-11-15:
I changed it to "ensenso_optical_frame" now. We only have a single frame for the camera images and the point cloud, so I guess that is a reasonable name that is similar to what other cameras use.

closed 2017-11-15 by saierd

gavanderhoorn, 2017-11-15:
Hm. Would it perhaps make sense to include the two frames anyway, but put an identity transform between them?

That would make it compatible to a lot of other depth sensors in ROS, while still modelling the fact that you only have a single frame.

gavanderhoorn, 2017-11-15:
Related: would you consider providing a `ensenso_description` (or `ensenso_support`, depending on how many products you intend to support) in combination with the driver?

That would allow you to standardise these frame/link names easily in a urdf, and remove the burden of modelling the camera from your users.

gavanderhoorn, 2017-11-15:
Captured this in #3.

commit ee9cb826 (2017-11-15): Change the default camera frame to "ensenso_optical_frame", fixes #2