ensenso/ros_driver #6: Starting cameranode takes longer than wait_for_server

mklomp, 2018-02-08:
When staring the camera-node it consistently takes about 22 seconds for the camera to be opened.

`[ INFO] [**1518085380.386794234**]: Initializing nodelet with 4 worker threads.
[ INFO] [**1518085402.302794783**]: Opened camera with serial number 'xxxxxx'.`

The wait_for_server in e.g. imagestream and request_data is set to 10 seconds. Therefore I get a timeout error when starting the camera-node and request_data from one launchfile (same holds for texture_pointcloud.launch)

saierd, 2018-02-09:
This is true. Some camera models can take quite long to open, so I increased the timeout.

closed 2018-02-09 by saierd

gavanderhoorn, 2018-02-09:
Would it be nice to make this a configuration parameter?

That way the timeout could be configurable by the user. I can imagine that for some cameras, it's either 10 seconds in which it should come up, or not at all. Having to wait 60 seconds for things to time out would introduce a lot of waiting.

saierd, 2018-02-09:
That makes sense, I added a parameter for changing the timeout.

I kept the long timeout by default, though. When the script is launched together with a camera node, it should wait. If something went wrong while opening the camera, the user will still get an error from the camera node before the timeout passed.

commit 5c7b1bf7 (2018-02-09): Increase the timeout in the utility scripts, fixes #6

When starting the scripts together with a camera node, the camera needs to be opened first. Depending on the model and the configuration, this can take some time.
commit aa2f899e (2018-02-09): Make the utility script timeout a parameter, fixes #6