ensenso/ros_driver #7: Add rectified depth image as output

sepjansen, 2018-02-19:
In addition to the pointcloud format, would it be possible to return a rectified depth image as well? Looking at the NxLib API, this would be similar to the single channel pointmap from the RenderPointMap function:
https://www.ensenso.com/manual/cmdrenderpointmap.htm

The reason is that many detections can be done on the depth image instead of a pointcloud. Now, we recreate the depth image from the pointcloud result ourselves, but it seems cleaner if the driver would expose this format directly.

Is this feature of interest to others as well?

saierd, 2018-02-19:
Sure, that's possible. I assume that such a depth image should be represented as a float image as it is described in [REP 118](http://www.ros.org/reps/rep-0118.html) and as it is used by [depth_image_proc](http://wiki.ros.org/depth_image_proc), right? If so, I will have a look into this soon.

sepjansen, 2018-02-19:
Correct, that's what I meant!

sepjansen, 2018-04-03:
Any progress on this? We would be interested in adding this functionality ourselves and make a PR if this fits with the design of the driver.

saierd, 2018-04-03:
I didn't get around to do it yet. If you want to implement it yourself I will gladly accept a PR.