ensenso/ros_driver #8: ensenso_camera_test tests are failing

reinzor, 2018-02-27:
```
[1519720055.786455757 ERROR /ensenso_camera_node]: Error while opening the camera 'ros_test'!
[1519720055.786535334 ERROR /ensenso_camera_node]: FileStereoCamera: Could not initialize background calibration thread
[1519720055.786575494 ERROR /ensenso_camera_node]: Failed to open the camera. Shutting down.
```

saierd, 2018-02-27:
This is because the tests use cameras from image files. Those are not officially supported in any version of the SDK that is currently available (and the error you got is a known problem in older versions).

They will be supported in the upcoming version 2.2, though, which we will hopefully publish in a few days.

closed 2018-02-27 by saierd

saierd, 2018-03-15:
FYI, we just published the new version 2.2 of the SDK, which can now run the tests.