ensenso/ros_driver #10: Issues with multi-camera setups

Danfoa, 2018-10-22:
In our setup we have two Ensenso N35 Cameras, pointing at the same workspace from different angles, but when using a multicamera setup, some questions (which I believe could be bugs) arise.

1. How to select the `ActionServer` to use. In the case when having two or more cameras, how does the `request_data` action chooses which action server to connect to and how do we change this?. 

As a matter of fact, if more than one Camera is open (more than one `ensenso_camera_node`s are running), when triggering the action request_data both cameras start publishing the data requested, but there is a pronblem...se number 2. 

2. Since the output topics of the camera nodes do not use the node' name as part of the `topic's names` when opening more than one camera all `ensenso_camera_node`s publish to exactly the same topics, which is in most cases not desired.

One way to get around this could be to add a node namespace, which I tried but breaks the functionality of the actionlib package, and I am also not so familiarized with the correct usages of node's namespaces. 

This can be solved by remapping the topics like so:
``` <remap from="/point_cloud" to="/lower_point_cloud"/> ```
But the `ensenso_camera_node` has so many output topics that this would not be efficient, your will end up remapping more than 20 topics.

Here is an example of the launch file I am ussing, although remaping the pointcloud topic works as desired I am not sure if the multi publisher problem described in 2 will create issues in further application development.

```
<launch>

  <arg name="upper_camera_serial" default="171864"/>
  <arg name="lower_camera_serial" default="171865"/>

  <node pkg="ensenso_camera" type="ensenso_camera_node" name="ensenso_upper_camera_node" output="screen">
    <param name="serial" value="$(arg upper_camera_serial)"/>
    <param name="camera_frame" value="ensenso_upper_optical_frame"/>
    <param name="target_frame" value="ensenso_upper_optical_frame"/>
    <remap from="/point_cloud" to="/point_cloud_upper"/>    
  </node>

  <node pkg="ensenso_camera" type="ensenso_camera_node" name="ensenso_lower_camera_node" output="screen">
    <param name="serial" value="$(arg lower_camera_serial)"/>
    <param name="camera_frame" value="ensenso_lower_optical_frame"/>
    <param name="target_frame" value="ensenso_lower_optical_frame"/>
    <remap from="/point_cloud" to="/point_cloud_lower"/>
  </node>

</launch>

```

saierd, 2018-10-23:
The intended way to solve this is to use namespaces. We currently don't provide a way to scope the topics explicitly, because that is exactly what namespaces are for. This shouldn't break the actions either, because they are just collections of topics, which all should get namespaced accordingly.

In your example, you should be able to start the nodes with the `ns` argument. For example, the node

    <node pkg="ensenso_camera" type="ensenso_camera_node" ns="upper_camera">
        ...
    </node>

will publish the topic `/upper_camera/point_cloud`. Its `request_data` action should also be available through an actionlib client on `/upper_camera/request_data`.

If the data of the two camera nodes gets processed separately, you can also use a `group` to start multiple nodes in the same namespace and then use the usual relative paths (e.g. `point_cloud` and `request_data`) in the nodes within this group.

Danfoa, 2018-10-23:
Thank you very much, I will have a try and see what mistake I made when I used namespaces.

Will update soon.

closed 2018-10-23 by Danfoa