ensenso/ros_driver #15: Camera does not support parameter Exposure

tabelh, 2019-03-07:
I am currently walking trying to complete the tutorial for the ROS ensenso driver. When i try to run my client, with the code provided http://wiki.ros.org/ensenso_driver/Tutorials/Parameters, i get the error that i am trying to use an attribute that is not available.
:~/ensenso_tutorials/ensenso_ws$ rosrun ensenso_camera ensenso_client.py 
Traceback (most recent call last):
  File "/home/thomas/ensenso_tutorials/ensenso_ws/src/ros_driver/ensenso_camera/src/ensenso_client.py", line 36, in <module>
    result = ensenso_client()
  File "/home/thomas/ensenso_tutorials/ensenso_ws/src/ros_driver/ensenso_camera/src/ensenso_client.py", line 25, in ensenso_client
    exposure = get_parameter.get_result().results[0].float_value
AttributeError: 'NoneType' object has no attribute 'results'

 this i could "solve" by deleting the last part (results & float_val) of the following statement: exposure = get_parameter.get_result().results[0].float_value

When i run the code it returns None, which I recon could be because I altered the last part of the statement above, but it also returns that the camera does not support the reading of the exposure parameter. 
[ WARN] [1551961309.027018767]: Reading the parameter Exposure, but the camera does not support it!
I added the bare minimum to the code provided in the tutorial. 
[ensenso_client.txt](https://github.com/ensenso/ros_driver/files/2941207/ensenso_client.txt)

saierd, 2019-03-11:
The message means that the current camera does not support the exposure setting (i.e. it doesn't have an [Exposure](https://www.ensenso.com/manual/index.html?_cameras_byserialno_$stereoserial_parameters_capture_exposure.htm) node in the NxLib tree).

What type of camera are you using? If you open it in NxView, can you adjust the exposure slider in the parameter window?

tabelh, 2019-03-13:
I am using a N35, to my memory adjusting the exposure in nxView was possible.

saierd, 2019-03-13:
That sounds weird, the N35 should definitely have the exposure node.

I found an error in your script, though. After sending the goal, you need to wait for the result to arrive with

```python
get_parameter.wait_for_result()
```

You can see a more complete example with waiting and error checking in the tests [here](https://github.com/ensenso/ros_driver/blob/ba8f2c9b534b198a42699a724ae9f48b97fc717c/ensenso_camera_test/test/parameter.py#L25).

Could you try if that fixes the problem? I don't know about the logged error message, though. That should not happen with your camera.

closed 2019-03-22 by saierd