ensenso/ros_driver #16: MapSMtoCores followed by core dump

tabelh, 2019-03-21:
I am trying to use an Ensenso N35 with ROS kinetic on ubuntu 16.04. When running the node as specified by the tutorial on my computer that has an nvidia gpu i get the following reponse 
```
Initializing nodelet with 16 worker thresds.
MapSMtoCores for SM 7.5 is undefined. Default to use 128 Cores/SM
Segmentation fault (core dumped)
```

`MapSMtoCores for SM 7.5 is undefined. Default to use 128 Cores/SM
`
is also stated when I open nxView. When I run the command using an ubuntu computer without an Nvidia gpu it runs fine. I am guessing there is an issue with the connection between the node and the gpu. I have been looking for the call to use the gpu in the code but can't find it.
Would you be able to provide a solution?

tabelh, 2019-03-21:
Sorry for the possible confusion. So i found where it went wrong. I am trying to link the ensenso camera to an image processing ros package. However when i change the name of a topic in camera.cpp the segmentation error occurs. I can not find a second call of the topic name and there don't understand the issue that is caused by renaming it. Have i overlooked a statement? How would i change the topic name where the image is published?

gavanderhoorn, 2019-03-21:
> How would i change the topic name where the image is published?

Use [remapping](http://wiki.ros.org/roslaunch/XML/remap)?

Never edit source code to change a topic name.

saierd, 2019-03-21:
First of all, the CUDA message for newer GPUs is known and fixed, but not in the current public version yet. It is not related to the segmentation fault, though.

I'm not sure why renaming a topic leads to a segmentation fault, that should be possible. But changing the code is not what you should do anyway. You should use the ROS [topic remapping](http://wiki.ros.org/Remapping%20Arguments) instead. When the two nodes rely on each other, you should put them into a launch file and then use the [remap node](http://wiki.ros.org/roslaunch/XML/remap) in the launch file XML to do this.

tabelh, 2019-03-21:
Thank you both for the quick reply. In that case I will use remapping.

closed 2019-03-22 by saierd

srini196, 2019-10-22:
I haven't changed topic names in the source. When i try to launch the default ensenso camera node however i receive the error: 

```
Initializing nodelet with 16 worker threads.
MapSMtoCores for SM 7.5 is undefined. Default to use 128 Cores/SM
[ERROR] [1571745829.016415388]: The camera '193941' is already in use
[ERROR] [1571745829.016442561]: Failed to open the camera. Shutting down.
```
I do not receive a segmentation fault, but the camera does not start and the nodelet shuts down.
 My Ubuntu 16.04 PC also includes an Nvidia 2080Ti GPU. However it is my understanding that the current ensenso ros driver does not support GPUs. I should still be allowed to start the camera nodelet as is, if i havent made any GPU specific modifications, correct?

On a Windows computer without a GPU, i am able to view the camera feed on NxView.

@saierd  Could you clarify what this means?
> First of all, the CUDA message for newer GPUs is known and fixed, but not in the current public version yet.

Could you guys provide any insight to help me fix this?

saierd, 2019-10-22:
The message 

```
MapSMtoCores for SM 7.5 is undefined. Default to use 128 Cores/SM
```

is a known bug in the 2.2 release of the NxLib and gets printed when using a GTX 20xx, which this version of the NxLib does not know yet. This is fixed in 2.3 (which was published in the meantime) but everything should still work (even computations on the GPU) in 2.2 even when this message gets printed.

----

The relevant error message here is

```
The camera '193941' is already in use
```

This means that some other process currently uses the camera (or at least the uEye driver thinks so). Do you have anything running, that could open the camera, e.g. an instance of NxView?

srini196, 2019-10-22:
Thanks for the clarification.
No other programs are accessing the camera. I've shutdown the windows system and restarted the Linux system, since i suspected some process was not getting killed. I continue to receive this error.

saierd, 2019-10-22:
Can you open the camera in NxView or the uEye camera manager on the Linux system?

srini196, 2019-10-22:
In the uEye Camera Manager app , the camera shows up as 
Free : yes
Available: no

saierd, 2019-10-22:
What does the uEye camera manager print at the bottom of the window when you click on the camera? Does it have a valid IP configuration?

If not, you should try to open the camera in NxView on Linux. It will automatically configure the IP for you.

srini196, 2019-10-22:
This was the problem:
> What does the uEye camera manager print at the bottom of the window when you click on the camera? Does it have a valid IP configuration?

This helped:
> If not, you should try to open the camera in NxView on Linux. It will automatically configure the IP for you.

Thank you !