ensenso/ros_driver #19: Ros driver cannot locate the pattern, while nxView can

natharos, 2019-04-02:
Dear reader, I hope you can help me solve this issue. I am trying to locate a Halcon pattern via ROS and project it in Rviz, so that I can use it later on for calibration. I have verified that the pattern is visible by locating it in nxView, where the axes are projected on the pattern. When trying this in ROS, the locatePattern action gives no feedback and no results. 

My launch file looks like:
```
<launch>
  <arg name="debug" default="false" />

  <arg unless="$(arg debug)" name="launch_prefix" value="" />

  <node name="ensenso_camera_node" pkg="ensenso_camera" type="ensenso_camera_node" respawn="false" launch-prefix="$(arg launch_prefix)" output="screen" />

  <node name="request_data" pkg="ensenso_camera" type="request_data" respawn="false" launch-prefix="$(arg launch_prefix)" output="screen" />

  <node name="static_transform_publisher" pkg="tf" type="static_transform_publisher" args="334.834 207.774 973.184 -0.704692 -0.379771 -0.599319 1.69578 map ensenso_optical_frame 100" launch-prefix="$(arg launch_prefix)" />

<node name="rviz" pkg="rviz" type="rviz" output="screen"  />
</launch>

```
Via the terminal I start the action with : 
```
rostopic pub /locate_pattern/goal ensenso_camera_msgs/LocatePatternActionGoal "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs: 0
  id: ''
goal:
  parameter_set: ''
  number_of_shots: 5
  target_frame: ''
  tf_frame: ''" 
```

What should I do differently?

gavanderhoorn, 2019-04-02:
Not an answer, but:

```xml
<node name="static_transform_publisher" pkg="tf" type="static_transform_publisher"
  args="334.834 207.774 973.184 -0.704692 -0.379771 -0.599319 1.69578 map ensenso_optical_frame 100 ..." />
```

The units of the first three elements of `args` are metres (`m`). Is your camera really 1 km up in the air?

natharos, 2019-04-02:
A fair point, I took them from the parameters (in mm) in NxView. Thank you for the tip.

saierd, 2019-04-02:
If you don't get any answer to your goal I suspect that it somehow doesn't arrive correctly. I never launched a goal from the terminal, so I don't know if your script is correct. Could you try to use the (pattern_marker)[http://wiki.ros.org/ensenso_driver/Nodes/pattern_marker] script and see if that works?

Please note that this ROS driver also uses meters to conform to the ROS conventions. You will therefore get different poses and point cloud data compared to the NxLib/NxView.

natharos, 2019-04-02:
I added `<node name="pattern_marker" pkg="ensenso_camera" type="pattern_marker" output="screen"  />` to the launch file, instead of trying to start the action via the terminal. Apart from the launch file (which contains only nodes from the ensenso_driver package) I do not use any other scripts.

`rostopic echo /locate_pattern/goal` gives messages like
```
header: 
  seq: 1336
  stamp: 
    secs: 1554212130
    nsecs: 512377023
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1554212130
    nsecs: 512340068
  id: "/pattern_marker-1337-1554212130.512"
goal: 
  parameter_set: "locate_pattern"
  number_of_shots: 0
  target_frame: ''
  tf_frame: ''
```

`rostopic hz /pattern_marker` keeps stating 'No new messages'

I have checked the rqt_graph and the locate_pattern action and the pattern_marker node are connected, so pattern_marker should be published...

saierd, 2019-04-02:
Are there messages on `/locate_pattern/result`? Does the camera node log any errors?

natharos, 2019-04-02:
locate_pattern does provide results, but the system does not find the pattern:
```
header: 
  seq: 96
  stamp: 
    secs: 1554213152
    nsecs: 386044725
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1554213151
      nsecs: 407064914
    id: "/pattern_marker-98-1554213151.407"
  status: 3
  text: ''
result: 
  found_pattern: False
  frame: ''
  pattern_poses: []
  patterns: []
  error: 
    code: 0
    message: ''
```
In the log files I could not find any error. 

Are there perhaps any settings or parameters that I have to change to get the same behaviour for the locate_pattern action as in nxView?

saierd, 2019-04-02:
The ROS node has a few limitations when it comes to collecting patterns. It can only collect stereo patterns that are visible in both cameras. And it always decodes the pattern data, so the pattern must contain data and it must be big enough in the images to be decoded. Did you enable "Specify Pattern Data Manually" in NxView?

natharos, 2019-04-02:
I do use the "Specify Pattern Data Manually" option to set my grid spacing of my Halcon pattern to 25mm, but both when this option is enabled and disabled nxView is able to find the pattern immediately in both images.

saierd, 2019-04-03:
So your pattern is not from us, but from Halcon and it doesn't have any white points like in [this image](https://www.ensenso.com/manual/index.html?howto_calibrationpatterns.htm), right?

Such a pattern is currently not supported in the ROS node, because it doesn't contain any information about its size and there is currently no way to specify the size manually for the ROS node. We should probably extend the action at some point to allow this. NxView will show the pattern in the overlay image even if you have the manual specification disabled, but you will notice that it doesn't compute its pose in that case (there is no 3D coordinate system drawn in the center of the pattern).

If your measurement doesn't need to be too precise, you can try to use NxCalTab (which was installed with the Ensenso SDK) to print a pattern with encoding yourself. The quality of the resulting measurement highly depends on your printer and how planar you can get the paper, though.

natharos, 2019-04-03:
With the pattern from NxCalTab it works! Thank you!

closed 2019-04-03 by natharos