ensenso/ros_driver #21: Ueye

ThomHordijk, 2019-04-17:
Currently we are trying to use an Ensenso n35. Up untill recently it worked just fine. 
Today the camera started throwing errors. We are not using custom code simply the ensenso_camera_node and request_data. When trying to start the camera nodes we sometimes get the following error, which we have not run into up until now:
```
[ERROR] [1555504175.323337616]: NxLibException 17 (ExecutionFailed) for item /Execute
[ERROR] [1555504175.323476283]: UEyeApiException: UEye API error 17(The driver could not be loaded) at </common/uEye/uEyeCamera.cpp:472>.
```
At times the camera does manage to start up and data is received however after some time Ros will lose the connection and a similar error message is shown. Could this pertain to the ueye driver or the SDK?

saierd, 2019-04-17:
This sounds like an issue with the uEye driver. Can `ueyedemo` use the camera reliably?

ThomHordijk, 2019-04-18:
The ueyedemo seems to working fine. The camera has not given the error the last couple of times that we tested with it. However, this seems kind of unreliable in regard to deploying the system on site. It should not have errors that come and go with the weather in my opinion

saierd, 2019-04-18:
I guess the problem also occurs in NxView, not only with the ROS node, right? If so, there is probably an issue with the uEye driver. You could try to reinstall it. If that doesn't help, you should contact the [IDS support](https://en.ids-imaging.com/support.html). They also provide support for our cameras and SDK.

closed 2019-04-23 by saierd