ensenso/ros_driver #28: How to set param with c++ ?

CaffeMrDe, 2019-10-22:
I saw the code in the ros wiki.
I understand the code about Setting param in camear with python. But , I could't set param with c++? I want to set the MINIMUM_DISPARITY .

```
# Add a new parameter set "location1". It will be initialized with the default
# settings from when the camera node started. We then change the exposure settings.
set_parameter.send_goal(SetParameterGoal(parameter_set="location1", parameters=[
    Parameter(key=Parameter.AUTO_EXPOSURE, bool_value=False),
    Parameter(key=Parameter.EXPOSURE, float_value=10)
]))
```

Could you show the tutriols with c++? Please, Thank you!

saierd, 2019-10-22:
The actions in this package are all based on [`actionlib`](http://wiki.ros.org/actionlib). The code that you copied from the wiki is just how you call such an action in Python.

To invoke an action from C++ you should follow the [actionlib tutorials](http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionClient). You have to create a client for the action type (in this case it would be an object of type `actionlib::SimpleActionClient<ensenso_camera_msgs::SetParameterAction>`) and use it to send an `ensenso_camera_msgs::SetParameterGoal` object to the action.

CaffeMrDe, 2019-10-22:
I have already done it. Precisely ，I code it flowllowing it. But it seem not work. 
```
    EnsensoSetting = new actionlib::SimpleActionClient<ensenso_camera_msgs::SetParameterAction>("set_parameter", true); 
    ensenso_camera_msgs::SetParameterGoal setParamGoal;
    setParamGoal.parameter_set = "MINIMUM_DISPARITY = -120";  // it is true? 
    setParamGoal.parameters = "";
    EnsensoActionclient->sendGoal(goal);
 
    EnsensoActionclient->waitForResult(ros::Duration(2.0));

    if (EnsensoActionclient->getState() == actionlib::SimpleClientGoalState::SUCCEEDED){
      printf("Yay! set init ok\n");
      return 0;
    }
```

saierd, 2019-10-22:
The way you filled the goal message is wrong. See the definition of the [SetParameter](http://docs.ros.org/api/ensenso_camera_msgs/html/action/SetParameter.html) action. The field `parameter_set` is for using multiple different sets of parameters.

You want to use the `parameters` field. In the generated code from the action message definition, this has the type `std::vector<ensenso_camera_msgs::Parameter>`, so each entry is a [Parameter](http://docs.ros.org/api/ensenso_camera_msgs/html/msg/Parameter.html) message. This is where you want to put the values that should be written to the camera.

In your case would look something like this:

```cpp
ensenso_camera_msgs::SetParameterGoal goal;
goal.parameters.resize(1);
goal.parameters[0].key = ensenso_camera_msgs::Parameter::MINIMUM_DISPARITY;
goal.parameters[0].float_value = -120;
```

and then send the `goal` object like you already did.

CaffeMrDe, 2019-10-22:
Thank you!  The problem is already solved!

closed 2019-10-22 by CaffeMrDe