ensenso/ros_driver #29: If i finish the handeye calibration, how to change the camera link?

CaffeMrDe, 2019-10-26:
I already finish the handeye calibration and get params about link. And how to put the cameraInBase frame in the NxLib rom.  And I want to recall the handeye Link, how can I do? 

Please tell me about something about it. Thank you!

saierd, 2019-10-28:
After you performed a hand-eye calibration, the resulting link gets automatically saved to the camera's EEPROM and will be applied to the 3D data. You shouldn't need to do anything else.

If you want to find out the offset that was calibrated by the hand-eye calibration, you can either take it from the result of the `calibrate_hand_eye` action or read it from TF, where the camera links gets published automatically. The exact TF frames depend on how you set up the camera node's frames. See the [camera frames tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/CameraFrames) for information on the different frames that the camera node uses.

CaffeMrDe, 2019-11-04:
Ok， I see. Thanks!

saierd, 2019-11-04:
Closing this for now, feel free to reopen if it still doesn't work.

closed 2019-11-04 by saierd