ensenso/ros_driver #32: High Calibration Errors

srini196, 2019-11-07:
I'm trying to perform Hand Eye Calibration with a fixed Ensenso N35 camera and a Yaskawa robot. Having ensured that the calibration poses are clearly visible to the camera, i proceeded using the hand eye calibration action. 

However i am getting errors in the order of 0.5 to 0.8 meters away from the actual position. The residual parameter is reported within 0.8 and 1.4. Correct me if i am wrong, but i understood the residual parameter to represent the error of the calibration with a value closer to 0 meaning better calibration.

 How do i improve the calibration quality?

yguenduez, 2019-11-07:
Hello srini186
Yes the residual error is the error per pixel. So less usually means a better calibration. In your case the average projected pixel is 0.8 to 1.4 pixels away from the real one.

Regarding the distance of 0.5 to 0.8 meters, that seems a bit far away. After the hand eye calibration: Does the link, stored in the camera node, comes close to the pose of the robot base? That could just be a bug in the transform we output in the action.

The camera point of origin lies in between the left sensor and the left lense. The exact point depends on the vergence angle of the sensor. So i cannot really tell. But if you want to "measure" the point of origin, you could do a workspace calibration. The result should be the Transformation form the Pattern origin to the camera point of origin (which, however, also depends on the intrinsic calibration of the camera).

In order to improve the quality of the hand eye calibration itself, we wrote a [guide](https://www.ensenso.com/manual/index.html?howto_handeyecalibration.htm) some time ago. The most frequent errors occur because of motion blur, or a moving robot during capturing images. In the latter case the position of the robot does not correspond to the pose in which the images were taken. So before you capture the images, move to a position and wait a while (1-2s) until the robot stops moving due to its inertia. Then capture images.

Please let me know if the stored link corresponds to the transform from camera to robot base or the other way around (inverse of the transform).

Best regards
Yasin

srini196, 2019-11-08:
The stored link does not correspond to either the transform or the inverse of the camera to robot base. The error is still in the range of 0.5 - 0.8m The calibration completed with the following stats:  

> used time: 2.168737  
> number of iterations: 206  
> residual: 1.323948  

Some additional information:

> camera_frame: ensenso_optical_frame  
> target_frame: base_link  
> fixed: true  
> wrist_frame: arm_left_link_tcp  

The camera is mounted at a fixed point. I'm trying to find its position relative to the robot base which is base_link in my case. The origin of base_link lies within the body of the robot.I decided not to use workspace calibration as I did not want to introduce human error while placing the calibration plate at an arbitrary position.

yguenduez, 2019-11-08:
Hello srini196

The frames you use seem to be correct.
Can you roughly describe where you put the pattern during the handeye calibration (on the robot wrist or somewhere fixed on the ground) and what kind pattern do you use (halcon or ensenso and size)? Did you print it yourself on paper or are you using a purchased pattern?

Best regards
Yasin

srini196, 2019-11-08:
The pattern is held by a gripper mounted on the end of the robot arm. I use an Ensenso 200 pattern with 24mm grid spacing which i have printed out on thick paper and pasted flat on a metal plate. I have had no problems finding the pattern so far.

yguenduez, 2019-11-08:
Hello srini196

Some days ago we have found a bug that is most likely be related to your case. If you use an Ensenso pattern, the hand eye calibration is currently broken, if both of the sensors do not see the whole pattern at the same time (with Ensenso patterns you usually only need to see a fraction of the pattern).
So what you could do is waiting for the next 2.3.x release of the EnsensoSDK, that is coming the next week, where this bug is fixed. Or simply use a Halcon pattern (there the whole pattern has to be seen by both sensors).
I think it's relatively hard to make sure that the Ensenso pattern will be seen by both sensors with a rotating and moving wrist without any feedback.

So maybe your issue could be fixed by just using another pattern type as a current workaround. The Halcon pattern can be found and saved in the NxCalTab application (delivered with the EnsensoSDK installation).

Best regards
Yasin

srini196, 2019-11-08:
@saierd mentioned in Issue #19 that the Halcon pattern can not be identified by the ROS driver. has this been fixed in a newer SDK version? I ask because i was not able to use the provided Halcon pattern for calibration with the driver; the pattern was not recognised.

> I think it's relatively hard to make sure that the Ensenso pattern will be seen by both sensors with a rotating and moving wrist without any feedback.

I have three sensors (to avoid blind spots) and it would be nearly impossible to have the pattern be visible to all of them simultaneously. I also assumed hand eye calibration should be used to calculate the position of individual cameras with respect to the robot base. Is there a workaround to calibrate three cameras at the same time or individually?

Currently i use the feedback from the CapturePattern action to check if the `found_pattern` parameter is true. Did you mean some other kind of feedback?

saierd, 2019-11-08:
The comment in #19 was for a pattern sold by MVTec. We also have patterns that use the same layout, but in contrast to the MVTec patterns, they have their size encoded in the points with a white center. These are also referred to as "Halcon" patterns in the NxLib and in NxCalTab. Any pattern that is sold by us/IDS or printed from NxCalTab should work in the NxLib and the ROS node.

You can link the three cameras together using the calibration wizard in NxView. See [this topic in our manual](https://www.ensenso.com/manual/index.html?multicamerasetups.htm) for more information. You can then link the root of your camera tree to the robot using a single hand-eye calibration.

---

The `found_pattern` flag is the correct way to check whether a pattern was found. The problem is that it does not indicate whether your Ensenso pattern was seen completely or not (it is enough to see a small patch of an Ensenso pattern to decode it and use it for a calibration). As Yasin already said, the current SDK releases have a bug, which causes the hand-eye calibration to fail when an Ensenso pattern was observed incompletely.

You therefore have the following options:

- Make sure that the camera can see the whole pattern. This is difficult, because the flag doesn't show this case and the ROS interface doesn't provide any additional information.
- Use NxCalTab to generate a pattern of type "Halcon". These can only be collected when they are fully visible, so the `found_pattern` flag will indicate the correct thing for them.
- We will probably publish a bugfix release of the Ensenso SDK next week, which will include a fix for this problem. When you use this, the calibration should also work with your Ensenso pattern.

srini196, 2019-11-11:
I will wait for the SDK update this week. In the meantime I'm facing a problem here. 

> You can link the three cameras together using the calibration wizard in NxView. See this topic in our manual for more information. You can then link the root of your camera tree to the robot using a single hand-eye calibration.

I position the Ensenso pattern in view of both cameras. In the nxView tracking section ,both cameras recognise the pattern. However when i try to calibrate the link between both the cameras at the same position, the software throws an exception "CaptureTimeout. Image capture timed out"

yguenduez, 2019-11-11:
Hello srini196,
to you point :
> However when i try to calibrate the link between both the cameras at the same position, the software throws an exception "CaptureTimeout. Image capture timed out"

When running the link calibration in nxView, there are two popup windows. One for defining the root camera in the camera "TF-like" tree and one window for defining the trigger mechanism.
In the trigger mechanism window you have to choose (via the serial number) a (master) camera, which sends a trigger signal to the other (slave) camera(s). This is used for synchronizing image capturing.
Did you choose a camera here, or did you click on the '**Software**' button? Selecting a camera here will lead to capture timeouts, if no trigger cable between the cameras is actually used.

srini196, 2019-11-11:
I was choosing a camera earlier. Choosing software allowed me to calibrate the relative poses of the cameras.

yguenduez, 2019-11-12:
Hello srini196.

The [xacros](https://github.com/ensenso/ros_driver/tree/master/ensenso_description) we offer currently use a mounting tf frame, the location where the camera is mounted (the center of the 4 screws). So you would have to use an offset transformation yourself from the left lenses of the cameras to the mounting centres, in order to use our URDF xacros properly. I guess we will update them soon (or adjust them to the camera's optical frame, which is the camera's origin in the left lens).

If you just want to have the tf tree with cameras linked to each other, you could use a [launch-file](https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/launch/mono_stereo_nodelets.launch), where you have to set a `link_frame` (In this particular example, you would have to exchange  the mono camera for stereo camera of course, e.g. exchange `nodelet` for `nodelet_mono`). The logic of the resulting TF Tree is described [here](http://wiki.ros.org/ensenso_driver/Tutorials/CameraFramesMultiple). The Camera will look for the stored link (transform) from the link calibration and will publish this transform relative to the tf Frame given in the parameter `link_frame`. This way you'll get a tf tree with one root camera and several other cameras linked to each other *relative* to one another.

So to come back to your question:

> What changes would have to be made to the camera launch file in order to have all cameras display the pointclouds in the same coordinate frame in Rviz?

If you use a the launch file described above with several nodelets using the `link_frame`, they all will publish their point clouds to the `link_frame` coordinate system. Because you now have a valid camera tf tree, you also have all the point clouds available in the `target_frame` of the root camera.

**Note**: In order to use the calibrated links, you will have to use nodelets instead of nodes, which is described [here](http://wiki.ros.org/ensenso_driver/Tutorials/MultiCamera).

yguenduez, 2019-12-06:
Hello srini196,

as with the new EnsensoSDK release ([here](https://www.ensenso.com/de/support/sdk-download/)), the hand eye calibration bug with the Ensenso Pattern is fixed.

And with the other issues you mentioned here, do they still exist?
I added some launch files examples in the tutorials [here](http://wiki.ros.org/ensenso_driver/Tutorials/CameraFrames) and also added a minimalist python example for a moving hand eye calibration [here](http://wiki.ros.org/ensenso_driver/Tutorials/HandEyeCalibration) (the fixed hand eye calibration is still in progress though).
Because our URDF/Xacro files do not match our current ensenso_camera package, they are currently being reworked and there will also be a tutorial on how to use them properly.

Best regards
Yasin

yguenduez, 2019-12-10:
Hello srini196,

I also added a little example with a tutorial on how to use our xacros, if you are still interested. They are written down [here](http://wiki.ros.org/ensenso_driver/Tutorials/EnsensoUrdf).

Best regards
Yasin

yguenduez, 2019-12-16:
Hello srini196,

I will close this issue. If any other errors occur, feel free to open a new ticket.

Best regards
Yasin

closed 2019-12-16 by yguenduez