ensenso/ros_driver #33: Line-ending issue in camera.cpp?

gavanderhoorn, 2019-11-15:
Not sure whether it was intentional, but c549c808 seems to have done some bad things with the line-endings in `camera.cpp`.

gavanderhoorn, 2019-11-15:
Related: would there be interest in setting up some form of CI in this repository?

yguenduez, 2019-11-15:
Thanks for the hint! The line ending was not intended. And 

> interest in setting up some form of CI in this repository?

Of Couse. This is probably the next thing to do.

gavanderhoorn, 2019-11-15:
> > interest in setting up some form of CI in this repository?
> 
> Of Couse. This is probably the next thing to do.

Could I suggest using [industrial_ci](https://github.com/ros-industrial/industrial_ci) for this?

The only 'difficult' bit would be the sdk, but seeing as you are maintaining that yourself, it should be more than possible to configure things such that it can be installed into the container before performing the rest of the build.

If using `industrial_ci`, I'd recommend using the [master branch](https://github.com/ros-industrial/industrial_ci/tree/master), as it's received quite a bit of attention since `legacy`.

yguenduez, 2019-11-15:
Thanks! I will have definitely have a look into it. And ..
..how do you respond that fast? :)

gavanderhoorn, 2019-11-15:
Github has had email notifications for quite a while now ;)

gavanderhoorn, 2019-11-15:
Perhaps @ipa-mdl could comment on what the best/most efficient way would be to get a non-`apt`/non-ROS/non-`rosdep`pable pkg dependency injected into an `industrial_ci` Travis setup.

closed 2019-11-15 by yguenduez

gavanderhoorn, 2019-11-15:
See #34 for a proposed CI configuration.

commit 6131c005 (2019-11-15): Fix broken line ending, fixes #33