ensenso/ros_driver #36: Running camera node without specifying serial causes crash

jornb, 2020-01-09:
When I run the camera node (or nodelet) without specifying the serial number, `rosrun ensenso_camera ensenso_camera_node`, I get a crash (segmentation fault)

Looking at the code, the error starts [here](https://github.com/ensenso/ros_driver/blob/4e02f8a3bfe855d014dd3bb0d629d78465d96cd0/ensenso_camera/src/nodelet.cpp#L48-L53):

```cpp
// No serial specified. Try to use the first camera in the tree.
NxLibItem cameras = NxLibItem()[itmCameras][itmBySerialNo];
if (cameras.count() > 0)
{
   serial = cameras[0].name();
}
```

In my item tree, I have no `itmBySerialNo`:
[image]
and so `serial` is actually assigned to `byEepromId`. The crash then happens [here](https://github.com/ensenso/ros_driver/blob/4e02f8a3bfe855d014dd3bb0d629d78465d96cd0/ensenso_camera/src/nodelet.cpp#L68):
```cpp
std::string type = NxLibItem()[itmCameras][itmBySerialNo][serial][itmType].asString();
```

Perhaps this the `BySerialNo` was removed in the upgrade to 2.3?

### Workaround

Specify the `serial` parameter explicitly, don't rely on auto-selection.

### System
* Ubuntu 18.03
* Ensenso SDK 2.3.1174

jornb, 2020-01-09:
On a separate note, hats off to the Ensenso team for creating this package and open sourcing it! Really helpful for us! 👍 ❤️ 🎉

yguenduez, 2020-01-09:
Hello jornb,

Thanks for your feedback and also the debugging, I really appreciate it!

Yes, since the 2.3.x versions of the EnsensoSDK we removed the `itmBySerialNo` constant. If you use it, it should still work. With the 2.3. Version this is now just a link to the `itmCameras` node. 
So if you use 
```cpp
NxLibItem cameras = NxLibItem()[itmCameras][itmBySerialNo];
````
it is the same as 
```cpp
NxLibItem cameras = NxLibItem()[itmCameras]
````
.

The crash itself happens, because in the Cameras node the ByEepromId subnode comes first in alphabetical order. So the ByEepromId is accessed, which does not contain any children nodes accessed by 
```cpp
std::string type = NxLibItem()[itmCameras][itmBySerialNo][serial][itmType].asString();
```
and throws an NxLibException, which is not handled currently.

I want to emphasise that I really love your kind of ticket as it makes debugging very easy for us!

Best Regards
Yasin

closed 2020-01-09 by yguenduez

jornb, 2020-01-09:
Thanks for the quick fix!

commit d3755ba0 (2020-01-09): Opening first appropriate camera if no serial is given, fixes #36