ensenso/ros_driver #44: about the rgbd

ghx0x0, 2020-07-21:
hi, 
   I 've found that you were developing the feature for the   rgbd-data-request since 2017, but it closed till May in Issues    #Feature/request depth rgb in 1#  
   Is this stopped?  I think this feature is important.

    By the way, I've tried code in this pull request, but found dependency:
           <depend>rgbd</depend>
           <depend>sr_foundation</depend> 
       in the  ensenso_camera/package.xml,   I could not find these packages  in ros,  can anybody tell me where to find the  'rgbd' and 'sr_foundation' ?

     Thanks!!

saierd, 2020-07-21:
The pull request you found was code created by a user of this package and the pull request was probably opened accidentally. I guess the missing dependencies are custom packages from that user.

We already support mono camera and textured point clouds. See the tutorials in the ROS wiki for more information, in particular [this one](http://wiki.ros.org/ensenso_driver/Tutorials/TexturedPointCloud).

Is this what you need? If not, what data would you expect that this package cannot produce?

yguenduez, 2020-12-22:
Feel free to reopen the issue, if you have further questions.

closed 2020-12-22 by yguenduez