ensenso/ros_driver #51: Request for tags

diogo-sr, 2020-11-27:
Hi,

At this moment there are no tags nor releases made for this repo, making it difficult to be dependent on specific versions (requires commit tags, which is not ideal).

Is it an option start tagging further relevant releases/updates and (if possible) tag previous ones?

gavanderhoorn, 2020-11-27:
A complete *me too* comment, but just wanted to give my :+1: a bit more weight.

yguenduez, 2020-11-30:
Hello guys,
thanks for the heads up. We will add git tags to this repository in a [semantic-versioning](https://semver.org/) fashion, if that is okay for you.
That means:

- Breaking changes will get an increase in the major version number. 
- Backwards compatible features will get an increase in the minor version number. 
- Bugfixes and patches to these features will generally get an increment in the patch version number. 

All in all as major.minor.patch. We also try to tag the already released ones.

diogo-sr, 2020-11-30:
Dear @yguenduez ,

Yes, semantic versioning sounds good! Thanks a lot for taking the feedback into account.
Adding the tag to the previous releases too would be perfect. 
(I don't really thing so, but) If you need any help let me know.

Regards,
Diogo

yguenduez, 2020-11-30:
I've added tags to the already released commits now in a semantic version way and will keep that for the future. Again, thanks for your constructive contributions.

Best Regards,
Yasin

closed 2020-11-30 by yguenduez