ensenso/ros_driver #53: NxLibException 17 (ExecutionFailed)

daenny, 2021-02-03:
We are running tests with an ensenso n35 camera.

Randomly, the following error occurs: 
```
[1612358420.785077658 ERROR /camera_left/ensenso_camera]: NxLibException 17 (ExecutionFailed) for item /Execute/182467
[1612358420.785171708 ERROR /camera_left/ensenso_camera]: CaptureTimeout: Waiting for images timed out.
```

We are running on the latest SDK3.0311 with ueye 4.93 and the latest version of this repository

What can be the reason causing this issue?

saierd, 2021-02-04:
The error just means that the image from the camera did not arrive. This can have many causes, the most likely one is that there are problems with the network connection. Ideally, the camera is connected directly to your network adapter without any switches / other devices on the same switch. You can also try if the [tips on this manual page](https://www.ensenso.com/manual/3.0/about/performance.html) help.

If this doesn't help, please check whether you have the same problems when opening the camera in NxView. If you do the problem is not specific to the ROS driver. In that case please contact the [support at IDS](https://en.ids-imaging.com/support-ticket.html). They will help you to analyze the problem further.

PS: I thought your account name looked familiar and it turns out that I installed your climate_group project just a few weeks ago. The world is very small :smile:

daenny, 2021-02-04:
Haha, the world is small indeed! I hope the component works for you. It works well for my setup, but I see some issues for different use-cases.

Thank you for you analysis. The problem is that we encounter it only very sparsely. We are running endurance tests with a robot which takes about one image every 5-6 seconds. Sometimes it fails after 5 hours, sometimes after 20 hours. 

We have two different setups, one system has one ensenso directly connected to the PC and another setup has two ensenso cameras connected via a switch. The error happens on both systems. 

Here are the stats of the network card for yesterdays run on the system with only one ensenso. It failed after 8 hours of running.

```
        RX packets 50005492  bytes 65537447298 (65.5 GB)
        RX errors 0  dropped 0  overruns 2155  frame 0
        TX packets 6965009  bytes 572722667 (572.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

I do see that the network card is set to mtu 1500, while I thought we already had increased that to 9000, but that might have been on the other system. I will increase the value for this system again.

saierd, 2021-02-04:
Yes, the component works flawlessly for me. It's a simple setup, so I didn't see any drawbacks yet.

---

Your receive buffer is set to the maximum size already? This is also described on the page I linked, but the exact commands to change it might depend on your system. `overruns 2155` indicates that some packets were dropped because the buffer was full. This might be fine if it doesn't happen too often. The uEye driver will request a resend of the missing packets and the frame might still arrive within the timeout.

Some more options you can try:

- Increase the timeout. By default we try to compute a reasonable timeout automatically, but you can specify your own timeout with the `capture_timeout` parameter of the camera node.
- Catch the error and simply try again. If the network generally works well, the next frame will probably work again.

diogo-sr, 2021-02-04:
Hi. I work together with @daenny and this error also happened with previous versions of the driver and of the sdk (2.2.175). The `capture_timeout` was set to 10 seconds and the error still persisted. Moreover, once this happened the driver would not be able to capture more images, only after a full restart.

saierd, 2021-02-04:
Ok, that doesn't sound like network problems then. If increasing the receive buffer size doesn't change anything, I guess that this is a problem with the uEye driver.

Do you use the [uEye driver version 4.93 from our website](https://www.ensenso.com/support/sdk-download/)? If not you can try to update it. Otherwise you should contact the IDS support [here](https://en.ids-imaging.com/support-ticket.html). Link them to this ticket so that they can see that we already tried to fix the usual network-related problems.

daenny, 2021-02-16:
A short update:
We updated the MTU to 9000 on the systems and the error still occurs. However, we are running most of the systems on uEye 4.90, which due to some parts in our system which rely on features that we have in our (outdated) fork. We currently in the progress to update everything to the latest upstream driver, as everything we need seems to be available. We will keep you updated.

saierd, 2021-03-12:
I am closing this for now. Feel free to reopen if there are any questions left.

closed 2021-03-12 by saierd

mmoerdijk, 2022-10-20:
Hi, 

Another colleague from smart robotics here.  This error still occurs at our systems even with all the network settings at the recommended settings. The frequency of the errors however varies, so we do think external factors are at play, and we are currently investigating those. 

However I think there is also a bug in the driver.  At the occurrence of the timeout, the driver prints this error message, as expected.
```
[ERROR] [1666246534.266442441]: NxLibException 17 (ExecutionFailed) for item /Execute/4103216064
[ERROR] [1666246534.266522618]: CaptureTimeout: Image capture timed out.
```
But after that error, it is no longer possible to query the driver for a new image. Doing so results in the following error:
```
[ERROR] [1666247083.615673777]: NxLibException 17 (ExecutionFailed) for item /Execute/182311
[ERROR] [1666247083.615717851]: ParametersInvalid: The specified serial number '4103216064' does not match any open camera.
```
Looking with nxView the camera still shows up as in use, even after more than 5 minutes have passed.

I think it would be quite easy to reproduce this on your side by setting the before mentioned `capture_timeout` to a low value.

mmoerdijk, 2022-10-20:
@saierd since I did not open this issue im not able to reopen it.

saierd, 2022-10-24:
Hi,

the error `ParametersInvalid: The specified serial number '4103216064' does not match any open camera.` is from the NxLib and indicates that the camera is no longer open. This could either be because it was closed or because it was disconnected by the uEye driver. The ROS driver never closes a camera intentionally (except when it gets terminated), so it was probably a disconnect in this case.

There is a new version 4.96 of the uEye driver [on our website](https://www.optonic.com/en/support/download/ensenso-sdk/), which fixed quite a few problems, especially on Linux. You could try if that solves the problem.

Otherwise, could you please contact the [IDS support](https://en.ids-imaging.com/support.html) and point them to the things we already tried in this ticket? They will help you with troubleshooting and have more possibilities to investigate the uEye driver.