ensenso/ros_driver #54: arg 'serial_stereo' is not defined.

kankanzheli, 2021-02-28:
I'm using ensenso N30-802-16-IR stereo 3D camera. For the launch file, I don't know the number of Serial of the Ensenso stereo camera and Serial of a mono camera supported by the NxLib. How to find this two number?

saierd, 2021-03-01:
The serial number is printed on a label on the device. You can also see it in NxView, when you open the camera there. See [this topic in the manual](https://www.ensenso.com/manual/3.0/about/camera.html) for more information on how to set up your camera.

kankanzheli, 2021-03-01:
Yes. I know how to get serial number of this camera.
For exanple, in the nodelet.launch file, when I use the "<arg name="camera" value="150580" />", I change the value to my camera's serial number. 
But when I want to use "color_point_cloud.launch", I also need to change the value of
    <arg name="serial_stereo" doc="Serial of the Ensenso stereo camera" />
   <arg name="serial_mono" doc="Serial of a mono camera supported by the NxLib" />
When I change both value into my serial number, the launch file is not work. It will show:
  RLException: arg 'serial_stereo' is not defined. 
  Arg xml is <arg name="serial_stereo" value="$(arg serial_stereo)"/>
 The traceback for the exception was written to the log file
So what should I do?

saierd, 2021-03-02:
Unfortunately I don't really understand the problem. The message `RLException: arg 'serial_stereo' is not defined.` is a ROS error message. Did you define this parameter when starting the launch file? Note that in `color_point_cloud.launch` there is currently no `value` for the `serial_stereo` argument. You have to specify it on the command line or when including it in another launch file. Alternatively you can modify the launch file and add `value="<YOUR SERIAL>"`.

For using this launch file, you should also have a color camera in addition to your stereo camera. It sounds like you only have a stereo camera.

kankanzheli, 2021-03-07:
So I looked at the tutorial. There is no existing code to visualize depth images. Am I right?

saierd, 2021-03-08:
What exactly are you searching? You can visualize all the data from this node in RViz.

saierd, 2021-03-12:
I am closing this for now. Feel free to reopen if there are any questions left.

closed 2021-03-12 by saierd