ensenso/ros_driver #55: How to get Aligned to color Depth Image

akashjinandra, 2021-03-02:
Hello I'm wondering what's the best methodology to get an aligned to color depth image. I noticed that this tutorial shows how to get a colored point cloud: http://wiki.ros.org/ensenso_driver/Tutorials/TexturedPointCloud. This is great, but lots of alogrithms rely on the RGB and Depth images being aligned to do processing on color and finding corresponding depth pixels to find real world coordinates based on that. 

Currently my rgb camera (uEYE 525 ) outputs color resolution of 1200 by 1600, and my n35 outputs 1024 by 1280. After talking with Philip from IDS, we found this: https://github.com/ensenso/ros_driver/blob/master/ensenso_camera_msgs/action/TelecentricProjection.action
 which allows the user to specify a frame, and have it transformed to that perspective. We chose the color frame, and it output a 1024 by 768 image, which still isn't the same resolution as the color. You can also specify the size, but that seem to just crop the image. and not actually align them. Also it doesn't output a camera_info topic much like the mono/color/camera_info or stereo/depth/camera_info. 

Is there a way to do what I'm asking? The reaslsense/ZED/ZIVID cameras I've used all provide this topic, although they have the big advantage of being integrated in one package, however we appreciate the flexibility that IDS and Ensenso provide with their cameras, so we are looking forward to using these. 

Thanks and let me know what you think.

saierd, 2021-03-02:
The `texture_point_cloud` action does what you want, but it only outputs the point cloud at the moment. This point cloud is structured and aligned with the pixels of the mono camera's undistorted images. You can use it together with the images on the `mono/color/` topic if you capture slow enough that synchronization of the two topics is not a problem.

I agree that the action should also output the texture image and the corresponding camera info somehow.

akashjinandra, 2021-03-02:
Is there anyway to get this point cloud into a depth image? also anyway to get it to output the camera info topic correctly for this depth image?

saierd, 2021-03-03:
No, that's not implemented in our node at the moment.

I did not try it, but you should be able to use the `depth_image_proc/register` node instead of the `texture_point_cloud` action. It does the same, but outputs a depth image instead of the point cloud.

saierd, 2021-03-12:
I am closing this for now. Feel free to reopen if there are any questions left.

closed 2021-03-12 by saierd