ensenso/ros_driver #56: Rising/Falling Edge Trigger failure

akashjinandra, 2021-03-02:
Hello, I have a uEYE 525 color camera and N35 depth camera. We are seeing an issue where the cameras aren't keeping the settings that were set through NX View. 

Specifically, we are seeing that when we set the trigger for the uEye camera (software/continuous/RisingEdge) it also somehow gets applied to ensenso camera. So through the NxView we can have the ensenso camera be in software trigger and the uEye camera be in RisingEdge Trigger. (Ideal scenerio)

When we use the ROS node with the uEye camera set to rising edge it crashes the node. We did some experiments and found that when we set the uEye to continuous trigger it somehow also changes the ensenso to continuous trigger (even if it's json parameter file says differently). 

We notice that if you set the uEye to rising/falling edge in NXview without the n35 also connected it also crashes. If we can't somehow sync this with the rising/falling edge we get the bluelight in our rgb image which isn't the best. Here is what our error is:
 
[ERROR] [1614298497.177103587]: NxLibException 17 (ExecutionFailed) for item /Execute/4103638864
[ERROR] [1614298497.177372969]: CaptureTimeout: Image capture timed out.
[ERROR] [1614298497.579554535]: NxLibException 17 (ExecutionFailed) for item /Execute/204774
[ERROR] [1614298497.579626236]: CaptureTimeout: Image capture timed out.
[WARN] [1614298497.602803]: Stereo request action was not successful.
[WARN] [1614298497.603948]: code: 17
message: "ExecutionFailed"
[WARN] [1614298497.605043]: Mono request action was not successful.
[WARN] [1614298497.605817]: code: 17
message: "ExecutionFailed"

saierd, 2021-03-02:
These error messages are expected when you set up a hardware trigger without connecting the camera which outputs the trigger signal. The message `CaptureTimeout: Image capture timed out.` means that no image arrived, because the camera did not receive a trigger signal in time.

If you use such a setup you have to make sure to trigger both cameras in time (with the `request_data` action). You should start it for the mono camera first to make sure it is ready when the trigger signal comes. Then you have to start it for the stereo camera within the timeout of the mono camera.

The default timeouts might not be long enough for a hardware trigger setup. You can specify it with the `capture_timeout` parameter of the nodes.

Unfortunately I don't have a setup to test the wrong parameter from the JSON file right now. Of course every camera should load the parameters from its JSON file if you specify them in the node parameters.

akashjinandra, 2021-03-02:
Hello, thanks for your response. We are using the node: https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/launch/color_point_cloud.launch which takes care of the action calling for us. But we notice that somehow even though we set the Stereo camera to Software trigger and Mono to continuous, somehow the stereo is also getting set as continuous as indicated by the speed of the light.

saierd, 2021-03-03:
- You did specify the Json files that you saved from NxView for both cameras in the node parameters, right?
- If you did, can you check during runtime whether they were loaded correctly? Unfortunately, there is currently no way to connect to the NxLib in the ROS node, but you can insert a call to `nxLibOpenTcpPort()` after the `nxLibInitialize` [here](https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/src/nodelet.cpp#L21) and then connect to it with NxTreeEdit to inspect the parameters. The trigger setting is in `/Cameras/<SERIAL>/Parameters/Capture/TriggerMode`.

akashjinandra, 2021-03-03:
Hello,

Thanks for your help on this.  We tried launching the mono camera first and that seemed to solve the issue.  We decided to use a timed roslaunch command to force the rgb camera to launch first and then the n35, it still seemed like it isn't super consistent on which node launches first.

 Again we are using the color point cloud python file, every once in a while the n35 will come up first causing the nodes to crash.  

Do you have any suggestions to have the cameras consistently come up in one launch file?

saierd, 2021-03-04:
With crashing you mean the error messages from your original post?

The order should not matter, this sounds like a bug somewhere. But I don't have any idea for what could cause this yet, so I can't really tell how you can work around it. We will investigate this.

pfreidin, 2021-03-11:
@akashjinandra Can this issue be closed?

akashjinandra, 2021-03-11:
Yes we can close this. We fixed it by attaching nodlet manager to mono camera. Thanks for the help and support.

closed 2021-03-11 by akashjinandra