ensenso/ros_driver #59: CameraInfo header timestamp is zero

thomascent, 2021-04-23:
Hi Ensenso,

I've been trying to register the ensenso's depth images to the frame of an RGB camera using a [depth_image_proc/register](http://wiki.ros.org/depth_image_proc?distro=noetic#depth_image_proc.2Fregister) nodelet. I was having some trouble because that nodelet uses a [message filter](http://wiki.ros.org/message_filters/ApproximateTime) to sychronise the color, depth and camera info messages it subscribes to and it turns out that the `camera_info.header.stamp` published by the ensenso driver isn't filled in. Here's one I `rostopic echo`ed just now, not to labor the point but you can see that `stamp: secs` and `stamp: nsecs` are both `0`

```
header: 
  seq: 1
  stamp: 
    secs: 0
    nsecs: 0
  frame_id: "ensenso_camera_frame"
height: 1024
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [1078.28662109375, 0.0, 566.381103515625, 0.0, 0.0, 1078.28662109375, 518.2259521484375, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 1
binning_y: 1
roi: 
  x_offset: 0
  y_offset: 0
  height: 1023
  width: 1279
  do_rectify: True
---
```
Any chance you could set the camera info timestamp to match the stamp attached to the depth image that's published alongside it? That seems to satisfy `depth_image_proc/register`

thomascent, 2021-05-03:
Any updates on this issue? I took a look at the source code just now and the problem seems to be that `leftRectifiedCameraInfo`'s stamp is only filled in when `goal->request_rectified_images` is `true` (which in my case it wasn't) here:
https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/stereo_camera.cpp#L284

But regardless `leftRectifiedCameraInfo` is still published alongside the depth image below:
https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/stereo_camera.cpp#L365 

It would be pretty straightforward to fix this by just adding the following:
```cpp
leftRectifiedCameraInfo->header.stamp = depthImage->header.stamp;
```
Immediately after this line:
https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/stereo_camera.cpp#L356

saierd, 2021-05-03:
Thank you for looking into this. Your suggestion looks correct, I will gladly accept a PR that makes this change.

closed 2021-05-05 by saierd