ensenso/ros_driver #60: Depth image units are floating point but expressed in millimeters

thomascent, 2021-04-23:
Hello again,

It seems that the depth images published by this package are encoded as 32FC1 but are expressed in millimeters. It states pretty explicitly in the remarks section that: 
```
All inputs and outputs of this package are in meters and seconds, as the convention for ROS requires
```
So I'm guessing this isn't the intended behavior

Just in case you need to reproduce this on your side, I'm on the latest master of `ros_driver` and I'm running the camera using the instructions straight out of the `Viewing Your First Point Cloud in RViz` tutorial on the wiki:

```
$ rosrun ensenso_ca ensenso_camera_node
$ rosrun ensenso_camera request_data
```
[Here's a bag](https://github.com/ensenso/ros_driver/files/6364396/depth_mm.zip) containing a couple of depth messages I collected, running them through this script prints `543.116` which sounds about right for the distance from the camera to the table it's looking at. 
```
import rosbag
from cv_bridge import CvBridge

bridge = CvBridge()

for topic, msg, t in rosbag.Bag('/home/ascent/Desktop/depth_mm.bag').read_messages():

    if topic == '/depth/image':
        depth_msg = msg
        depth = bridge.imgmsg_to_cv2(msg)

print(depth[800,400])
```
Is there a parameter somewhere that controls this? If not any chance you could publish the depth images in meters?

thomascent, 2021-05-03:
Any updates on this issue? Assuming this isn't the expected behavior then it should be pretty easy to fix this by just adding the following: 
```cpp
depthImage /= 1000.0;
```
immediately after this line:
https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/image_utilities.cpp#L174

saierd, 2021-05-03:
Thank you for looking into this yourself. The behavior is not intended, we want to convert everything to meters and this was probably an oversight when we added the depth image functionality. I will gladly accept a PR that changes this.

closed 2021-05-06 by saierd