ensenso/ros_driver #64: Port to ROS2?

cardboardcode, 2021-06-10:
Hi, thank you for providing and maintaining this ROS driver for Ensenso. 

May i ask if there are any plans to port this to ROS2? 😄 

If there is none, is there a way for people like me to start contributing to a port?

saierd, 2021-06-11:
There are no plans for ROS2 yet, since you are the first one to ask about it. Unfortunately we don't have anyone who can work on the ROS interface at the moment, but contributions are always welcome.

I don't know much about ROS2 yet, but some quick googling showed that it is probably not possible to keep the code compatible with both ROS 1 and 2, right? That is a bit unfortunate, since it forces us to maintain two versions of the code. As long as the differences are small enough, this could maybe happen in separate branches of this repository, so that we can share some patches between the two versions.

cardboardcode, 2021-06-11:
Hi @saierd, thanks for the prompt clarification. 

That is indeed unfortunate... Hmmm... :sweat_smile: 

I will fork this repository and attempt to create a simple ROS2 interface with an Ensenso N35 that I have. Not sure if I can achieve a desirable level of code robustness and quality as this repository's codebase but I will do my best.

Will update here once there is any progress on my side. Let's use this thread to keep track of any ROS2 porting effort for Ensenso.

cardboardcode, 2021-06-11:
Verified that **ROS2 Foxy** relies on **PCL** version `1.10.0`.

In this **PCL** version, there is an `ensenso_grabber.cpp` source file which would serve as a starting point for a ROS2 interface. 

**TODO**
- [ ]  Validate use of `ensenso_grabber.cpp` using **PCL**`1.10.0`in obtaining scene pointcloud from Ensenso N35.
- [ ]  Create ROS2 package that uses `ensenso_grabber.cpp` library functions, gets scene pointcloud, converts and publishes as `sensor_msgs/PointCloud2`
- [ ]  Populate meta-data for published pointclouds.
- [ ]  Elaborate ROS2 package to publish rectified image, depth image, infrared image and camera intrinsic parameters.
- [ ]  Optimize speed and accuracy of published pointclouds.
- [ ]  Implement GitHub Action CI workflow with integration to `industrial_ci`.
- [ ]  Adhere ROS2 package to REP 2004 Quality Level 4.

jornb, 2021-06-11:
I too would like a ROS2 port. But for others looking at this I have found that [ros1_bridge](https://index.ros.org/p/ros1_bridge/github-ros2-ros1_bridge/) works well for our use case. However, running ROS1+ROS2 has some limitations of course.

I'd be interested in helping with this effort during the fall of this year. I've used this driver before and made some changers to it in the past.

For backwards-compatibility, I suggest keeping the ROS message/action/service API the same when migrating to ROS2. I also suggest keeping the source code from the existing `ensenso_camera` package, as this API is much richer than just fetching point clouds, as the PCL `ensenso_grabber.cpp` does.

saierd, 2021-06-11:
Yes, I would prefer keeping the same API and sharing code for the interaction with the NxLib between the two versions.

As far as I could see it is necessary to change a lot of includes and C++ API for ROS2. That is probably too widespread to easily take care of with the preprocessor. Is there maybe a compatibility layer that allows to call some simple ROS APIs independently of the version?

cardboardcode, 2021-06-11:
> For backwards-compatibility, I suggest keeping the ROS message/action/service API the same when migrating to ROS2. I also suggest keeping the source code from the existing ensenso_camera package, as this API is much richer than just fetching point clouds, as the PCL ensenso_grabber.cpp does.

> Yes, I would prefer keeping the same API and sharing code for the interaction with the NxLib between the two versions.

@jornb @saierd Noted. Will try to keep the same API as ensenso_camera. As a contingent, I will keep the `ensenso_grabber.cpp` as a back-up plan. 

> As far as I could see it is necessary to change a lot of includes and C++ API for ROS2. That is probably too widespread to easily take care of with the preprocessor. Is there maybe a compatibility layer that allows to call some simple ROS APIs independently of the version?

From my experience developing with ROS2 and ROS1, we would have to create this compatibility layer from scratch. ROS2 uses a drastically different client library for basic CPP and Python implementation.

cardboardcode, 2021-06-30:
##  Investigate performance of running Ensenso as ROS1 package with ros1_bridge.
This comment will track the actions I took to achieve this todo task.

### Environment
1. Ubuntu 20.04
2. ROS1 Noetic
3. ros_driver (f31847ae9341ad03a0aaee13433d7f4d1a5de546)
4. ensenso-sdk/now 3.1.969

### Setup of Ensenso SDK
```bash
cd $HOME
wget https://download.optonic.com/s/ensensosdk/download?files=ensenso-sdk-3.1.969-x64.deb
sudo dpkg -i ensenso-sdk-3.1.969-x64.deb
```

### Setup of ROS1 driver
```bash
cd $HOME
mkdir -p ensenso_ros1_ws/src && cd ensenso_ros1_ws/src
git clone https://github.com/ensenso/ros_driver.git --single-branch --branch master --depth 1
cd ~/ensenso_ros1_ws
source /opt/ros/noetic/setup.bash
catkin build
```

cardboardcode, 2021-08-08:
## Explore use of `ensenso_viewer.cpp` using PCL `1.8.0` in **obtaining scene pointcloud** from Ensenso N35.

# Conclusion :stop_sign:
Using `ensenso_viewer.cpp` in PCL `1.8.1` outputs point cloud frames at **slight more than** `~2` **frames per second** (fps), as shown by the following terminal output:
```bash
FPS: 2.14427
FPS: 2.19127
FPS: 2.3174
FPS: 2.33683
FPS: 2.37841
FPS: 2.64333
FPS: 2.7017
FPS: 2.77804
FPS: 2.76929
FPS: 2.77141
``` 
### Environment
- Ubuntu **18.04**
- PCL `1.8.1`
- Ueye `4.93.0`
- Ensenso SDK `3.1.969`

### Setup
Follow the instructions below to reproduce the observed results in **Conclusion**.

#### Build PCL 1.8.1 from source.
This section is going to take about an hour to install. So go have a coffee or tea or something. :coffee:
```bash
cd $HOME
wget https://github.com/PointCloudLibrary/pcl/archive/pcl-1.8.1.tar.gz
tar -xf pcl-1.8.1.tar.gz
cd pcl-pcl-1.8.1 && mkdir build && cd build
cmake ..
make
```

#### Connect Ensenso N35
Instructions not shared for this since it depends on your physical setup.

### Run
Once you run the commands below, you should see the results observed in **Conclusion**.
```bash
cd $HOME/pcl-pcl-1.8.1/build/bin
./pcl_ensenso_viewer
```

saierd, 2021-08-09:
I have some comments regarding your tests.

- When using the `request_data` script without any parameters exactly 2 FPS are expected, since this script uses an [adjustable loop rate](https://github.com/ensenso/ros_driver/blob/dafd3ba54bcbf20f440785ab2d66f04452095edb/ensenso_camera/scripts/request_data#L11). This is not the maximum achievable frame rate.
- Measuring the frame rate depends a lot on your hardware, your camera model and the settings you use. In most cases, network transmission and stereo matching computation times inside of the NxLib will absolutely dominate your frame times and the influence of the library that you use to capture and copy the resulting point clouds (ROS/ROS2/PCL) will be very small.

benediktkreis, 2022-02-17:
Are there any updates? I saw @cardboardcode unofficial ROS2 driver with the last commit on 11/08/2021. Is it planned to release an official ROS2 driver in the near future? I am thinking of purchasing an Ensenso N35, but I can only consider ROS2 compatible cameras.

benediktkreis, 2022-05-09:
Are there any updates?

gavanderhoorn, 2022-06-24:
As may 9 was some time ago, just adding another 'me too' comment: would there be an ETA for perhaps a beta release?

benthie, 2022-06-24:
Hi everyone. I am happy to announce that we currently planning to release a beta in early July.

benthie, 2022-07-12:
Hi everyone and thank you for your patience. The wait is finally over: we just released v2.0.0-beta.

Please see the `README.md` file for documentation on how to build the ROS2 version of our package. If something is missing or not working, please let me know and either open an issue or just feel free to submit a PR.

benthie, 2022-08-08:
Has anyone of you had the time to test the `ros2` release yet? 

PS: We just released v2.0.0-beta.2 with a few minor fixes and xacro launch files.

gaudfra, 2022-12-13:
Hi, I'm finding some difficulties with the ROS2 release, both with Windows and Ubuntu: in Windows I am not able to run the prepare_ros2_build.sh file (actually I run it but it gives me the error: "mv: cannot stat 'ensenso_camera/package.xml'). With Ubuntu I am struggling with the colcon build at the end, which does not work properly, can someone help? Thanks

benthie, 2022-12-14:
Regarding your problem on Windows (I assume 10), I just tried the following with PowerShell and it worked as expected:
- go to a temporary directory and open a PowerShell
- `mkdir -p cws/src`
- `cd cws/src`
- `git clone https://github.com/ensenso/ros_driver.git ensenso_ros_driver`
- `cd ensenso_ros_driver`
- `git status` shows clean working tree
- `$env:ROS_VERSION=2`
- `.\.github\scripts\prepare_ros2_build.sh`
- `git status` shows a bunch of created and modified files

Regarding your problem on Ubuntu:
- which Ubuntu are you using?
- which ROS2 version are you using?
- what error messages do you get when you run `colcon build`?

gaudfra, 2022-12-14:
Hi @benthie, thank you so much for the reply. Yes, I'm using Windows 10: I followed your steps and the git status commands work, but after that when i try to use "colcon build" in the cws folder I get the following error: 
`PS C:\tmp\cws> colcon build
Starting >>> ensenso_camera_msgs
Starting >>> ensenso_description
--- stderr: ensenso_camera_msgs
Traceback (most recent call last):
  File "c:\python38\lib\site-packages\colcon_core\executor\__init__.py", line 91, in __call__
    rc = await self.task(*args, **kwargs)
  File "c:\python38\lib\site-packages\colcon_core\task\__init__.py", line 93, in __call__
    return await task_method(*args, **kwargs)
  File "c:\python38\lib\site-packages\colcon_ros\task\ament_cmake\build.py", line 59, in build
    rc = await extension.build(
  File "c:\python38\lib\site-packages\colcon_cmake\task\cmake\build.py", line 87, in build
    rc = await self._reconfigure(args, env)
  File "c:\python38\lib\site-packages\colcon_cmake\task\cmake\build.py", line 153, in _reconfigure
    raise RuntimeError(
RuntimeError: VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.
---
Failed   <<< ensenso_camera_msgs [0.62s, exited with code 1]
Aborted  <<< ensenso_description [0.59s]

Summary: 0 packages finished [1.14s]
  1 package failed: ensenso_camera_msgs
  1 package aborted: ensenso_description
  1 package had stderr output: ensenso_camera_msgs
  3 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
VisualStudioVersion is not set, please run within a Visual Studio Command Prompt.`
Before creating the directory I ran "& C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64". 
Where should I do the colcon build?

Regarding Ubuntu:

- Ubuntu 20.04.5

- ROS2 Foxy

- I get a long list of error (I don't know if it is convenient to attach it here), but the first ones are like: "In file included from /home/fragaud/ros2_ws/src/ros_driver/ensenso_camera2/include/ensenso_camera/ros2/senso_msgs/image_encodings.h:4, from /home/fragaud/ros2_ws/src/ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
7opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:47:12: error: conflicting declaration 'const char sensor_msgs::image_encodings::RGB8 []'
47 | const char RGB8[] = '"rgb8";
In file included from PATH, from PATH:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:47:23 note: previous declarations as 'const string sensor_msgs::image_encodings::RGB8'
47 | const std::string RGB8 = "rgb8";

benthie, 2022-12-14:
Regarding Windows 10:
- Does building other packages or let's say running the proposed [examples](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#try-some-examples) of the ROS2 installation for Windows work properly?
- > Where should I do the colcon build? 
    - In your case you should run it in `C:\temp\cws`, which is your overlay workspace
    - Please make sure to 1) source the underlay and 2) be in a Visual Studio environment as described [here](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#try-some-examples)

Regarding Ubuntu:
- Just to make sure: did you source the ROS environment (underlay) by running `source /opt/ros/foxy/setup.bash` (assuming that you installed ROS under the default install path `/opt`)?
- Could you run `rosdep install --from-paths src --ignore-src -r -y` from you workspace folder, e.g. from `~/temp/cws`, after you cloned the Ensenso ros_driver and after you ran `prepare_ros2_build.sh`?

gaudfra, 2022-12-14:
Let's just focus on Windows for the moment: I am able to run the talker-listener example, so I assume that ROS2 works properly. This is the steps I followed with the command prompt:

- Setup the environment: call "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Auxiliary\Build\vcvarsall.bat" x86_amd64
With the following output: 
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.21
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86_x64'

- Source the underlay: C:\Windows\System32>call C:\ros2-windows\local_setup.bat
- Run the colcon build in the tmp\cws folder (I already cloned the repo in the src folder) and this is the error now: 
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "cv_bridge", but CMake did not find one.

  Could not find a package configuration file provided by "cv_bridge" with
  any of the following names:

    cv_bridgeConfig.cmake
    cv_bridge-config.cmake

  Add the installation prefix of "cv_bridge" to CMAKE_PREFIX_PATH or set
  "cv_bridge_DIR" to a directory containing one of the above files.  If
  "cv_bridge" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< ensenso_camera [7.98s, exited with code 1]

Summary: 2 packages finished [2min 26s]
  1 package failed: ensenso_camera
  1 package had stderr output: ensenso_camera
  2 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)

Please tell me what I'm doing wrong

benthie, 2022-12-14:
To me this looks like you have a problem with your OpenCV installation. Did you follow the steps described [here](https://docs.ros.org/en/foxy/Installation/Windows-Install-Binary.html#install-opencv)? If OpenCV is installed and known to the ROS2 environment and you have sourced it, CMake should actually be able to find it.

If it would be okay for you, I'd like to focus on the Ubuntu problem since our CI currently only tests on Ubuntu. Testing on Windows is still a ToDo and I am currently setting up a Windows machine in order to test our ROS2 package there and until I am not done with that I can best help you with the Ubuntu issue.

gaudfra, 2022-12-14:
Yes that's not a problem, I'll try what you recommend before on Ubuntu and I'll let you know

gaudfra, 2022-12-15:
Hi @benthie, sorry for the late response. I installed ros2 under /opt, and I sourced it. I cloned the drivers successfully and I was able to run `prepare_ros2_build.sh`. After that, I ran `rosdep install --from-paths src --ignore-src -r -y` and the output was good (#All required rosdeps installed successfully). After that, I tried the colcon build but it failed just like yesterday.

This is the error:
fragaud@fg:~/ros2_ws$ colcon build
Starting >>> ensenso_camera_msgs
Starting >>> ensenso_description
Finished <<< ensenso_description [0.78s]                                
[Processing: ensenso_camera_msgs]                             
Finished <<< ensenso_camera_msgs [34.6s]                       
Starting >>> ensenso_camera
--- stderr: ensenso_camera                               
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:47:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::RGB8 []’
   47 | const char RGB8[] = "rgb8";
      |            ^~~~
In file included from /opt/ros/noetic/include/cv_bridge/cv_bridge.h:41,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:2:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:47:23: note: previous declaration as ‘const string sensor_msgs::image_encodings::RGB8’
   47 |     const std::string RGB8 = "rgb8";
      |                       ^~~~
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:48:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::RGBA8 []’
   48 | const char RGBA8[] = "rgba8";
      |            ^~~~~
In file included from /opt/ros/noetic/include/cv_bridge/cv_bridge.h:41,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:2:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:48:23: note: previous declaration as ‘const string sensor_msgs::image_encodings::RGBA8’
   48 |     const std::string RGBA8 = "rgba8";
      |                       ^~~~~
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:49:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::RGB16 []’
   49 | const char RGB16[] = "rgb16";
      |            ^~~~~
In file included from /opt/ros/noetic/include/cv_bridge/cv_bridge.h:41,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:2:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:49:23: note: previous declaration as ‘const string sensor_msgs::image_encodings::RGB16’
   49 |     const std::string RGB16 = "rgb16";
      |                       ^~~~~
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:50:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::RGBA16 []’
   50 | const char RGBA16[] = "rgba16";
      |            ^~~~~~
In file included from /opt/ros/noetic/include/cv_bridge/cv_bridge.h:41,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:2:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:50:23: note: previous declaration as ‘const string sensor_msgs::image_encodings::RGBA16’
   50 |     const std::string RGBA16 = "rgba16";
      |                       ^~~~~~
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:51:12: error: conflicting declaration ‘const char sensor_msgs::image_encodings::BGR8 []’
   51 | const char BGR8[] = "bgr8";
      |            ^~~~
In file included from /opt/ros/noetic/include/cv_bridge/cv_bridge.h:41,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:2:
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:51:23: note: previous declaration as ‘const string sensor_msgs::image_encodings::BGR8’
   51 |     const std::string BGR8 = "bgr8";
      |                       ^~~~
In file included from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/include/ensenso_camera/ros2/sensor_msgs/image_encodings.h:4,
                 from /home/fragaud/ros2_ws/src/ensenso_ros_driver/ensenso_camera2/src/image_utilities.cpp:12:
[... cut ...]
make[2]: *** [CMakeFiles/ensenso_camera.dir/build.make:76: CMakeFiles/ensenso_camera.dir/src/camera.cpp.o] Error 1
make[2]: *** [CMakeFiles/ensenso_camera.dir/build.make:180: CMakeFiles/ensenso_camera.dir/src/stereo_camera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:549: CMakeFiles/ensenso_camera.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ensenso_camera [23.7s, exited with code 2]

Summary: 2 packages finished [58.5s]
  1 package failed: ensenso_camera
  1 package had stderr output: ensenso_camera
  2 packages not processed

benthie, 2022-12-15:
Having a closer look on the error messages you posted yesterday, the following message
```
/opt/ros/foxy/include/sensor_msgs/image_encodings.hpp:47:12: error: conflicting declaration 'const char sensor_msgs::image_encodings::RGB8 []'
/opt/ros/noetic/include/sensor_msgs/image_encodings.h:47:23 note: previous declarations as 'const string sensor_msgs::image_encodings::RGB8'
```
tells us that you also have the ROS1 (noetic) underlay in your PATH variable. Could it be that you accidentally or automatically source the ROS1 underlay in your terminal? Anyways, by sourcing both ROS1 and ROS2 you get duplicate declarations.

benthie, 2022-12-15:
Having multiple ROS versions installed and even running them in parallel is possible as described [here](https://answers.ros.org/question/355548/can-i-run-ros2-and-ros1-on-one-ubuntu-at-the-same-time/), but you have to make sure that you only source one in each terminal.

If you open a fresh terminal and run e.g. `roslaunch` you should get an error saysing `Command 'roslaunch' not found ...`, which means that your terminal environment has no idea where to find this ROS specific command. If, on the other hand, this command runs without an error and says something like "Usage: roslaunch ..." then ROS1 is already sourced, which you do not want.

Regarding your EDIT:
If you change your bashrc you have to source it by running `source ~/.bashrc` if you are staying in the same terminal. If you open a new terminal, it automatically sources the new bashrc. You can test this with e.g. adding `alias sayhello="echo hello"` to your bashrc. If you can run `sayhello` in a terminal you know that it has sourced your bashrc.

gaudfra, 2022-12-15:
Right now if I open a fresh terminal I have the following messages:
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'foxy' before. Please make sure that the environment does not mix paths from different distributions.
ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.
not found: "/home/fragaud/ros2_ws/install/ensenso_camera/share/ensenso_camera/local_setup.bash"

If I run `roslaunch` I get: 
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/roslaunch", line 34, in <module>
    import roslaunch
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/__init__.py", line 62, in <module>
    from .scriptapi import ROSLaunch
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/scriptapi.py", line 42, in <module>
    import roslaunch.parent
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/parent.py", line 54, in <module>
    import roslaunch.server
  File "/opt/ros/noetic/lib/python3/dist-packages/roslaunch/server.py", line 80, in <module>
    from rosgraph_msgs.msg import Log
ImportError: cannot import name 'Log' from 'rosgraph_msgs.msg' (/opt/ros/foxy/lib/python3.8/site-packages/rosgraph_msgs/msg/__init__.py)

benthie, 2022-12-16:
If you are having such fundamental problems with your ROS environments I suggest to: 
- uninstall all ROS1 and ROS2 distros you have installed and cleanly reinstall them
- consider installing only ROS2 for the time being and get things working, then install ROS1 in case you need it
- check your bashrc and all other files that are executed at login by your bash (see e.g. [here](https://unix.stackexchange.com/questions/334382/find-out-what-scripts-are-being-run-by-bash-at-login) for one example of how to find out which scripts/files are executed/loaded at login of your bash) and make sure that you either do not automatically source any ROS environment or if so you only source the one you currently want
- check out some of the ROS examples and tutorials in order to make sure that your setup works fine
- consider [ROS Answers](https://answers.ros.org/questions/) for questions that are not directly related to our `ros_driver`, because a lot of questions have already been asked there and they probably provide better answers than I do ;)

But please: still feel free to ask your questions here if you're stuck and of course especially in case you encounter problems with our ROS2 driver. I wish you success!

gaudfra, 2022-12-17:
I managed to make it work: I reinstalled only ROS2 and even if I followed exactly the guide I had some packages missing (cv_bridge, image_geometry, image_transport, pcl_conversions, pcl_ros),  but after their installation the colcon build worked and I could use the ensenso_camera package. Thanks for the help!

benthie, 2022-12-17:
I am happy to hear that you managed to get it work. Feel free to open a new issue if you find any bugs or problems with the package and ROS2.