ensenso/ros_driver #66: Add support for S10

pfreidin, 2021-12-10:
This package only supports stereo cameras but the new S10 is a structured light camera. The following error occurs When attempting to launch an ensenso_camera/nodelet:
**The camera to be opened is of the wrong type StructuredLight. It should be Stereo.**

############### complete console output ####################

```
$ roslaunch ensenso_camera nodelet.launch
... logging to /home/tech/.ros/log/e09693b4-59f8-11ec-a68d-94c691a5898a/roslaunch-tech-NUC8i3BEH-14510.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tech-NUC8i3BEH:43197/

SUMMARY
========

PARAMETERS
 * /Ensenso_218148/serial: 218148
 * /Ensenso_218148/target_frame: Workspace
 * /rosdistro: melodic
 * /rosversion: 1.14.12

NODES
  /
    Ensenso_218148 (nodelet/nodelet)
    manager_ (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[manager_-1]: started with pid [14525]
process[Ensenso_218148-2]: started with pid [14526]
[ INFO] [1639169273.834979332]: Loading nodelet /Ensenso_218148 of type ensenso_camera/nodelet to manager /manager_ with the following remappings:
[ INFO] [1639169273.836988566]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [1639169273.845668463]: Initializing nodelet with 4 worker threads.
[ INFO] [1639169273.857829152]: waitForService: Service [/manager_/load_nodelet] is now available.
[ERROR] [1639169277.063881560]: The camera to be opened is of the wrong type StructuredLight. It should be Stereo.
[FATAL] [1639169277.174065622]: Failed to load nodelet '/Ensenso_218148` of type `ensenso_camera/nodelet` to manager `/manager_'
[manager_-1] process has died [pid 14525, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=manager_ __log:=/home/tech/.ros/log/e09693b4-59f8-11ec-a68d-94c691a5898a/manager_-1.log].
log file: /home/tech/.ros/log/e09693b4-59f8-11ec-a68d-94c691a5898a/manager_-1*.log
[Ensenso_218148-2] process has died [pid 14526, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load ensenso_camera/nodelet /manager_ __name:=Ensenso_218148 __log:=/home/tech/.ros/log/e09693b4-59f8-11ec-a68d-94c691a5898a/Ensenso_218148-2.log].
log file: /home/tech/.ros/log/e09693b4-59f8-11ec-a68d-94c691a5898a/Ensenso_218148-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

saierd, 2021-12-13:
We are currently working on support for S-series devices.

As a workaround, you might be able to get it to work already when you remove the error message [from the code here](https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/nodelet.cpp#L81) and keep in mind the following things:

* Explicitly specify the serial number to open.
* Do not request raw/rectified images or disparity maps.

The API for the point clouds should be compatible with other devices, so this might just work.

saierd, 2022-02-03:
Done with #71

closed 2022-02-03 by saierd