ensenso/ros_driver #67: Linked camera calibrtion in eeprom

sepjansen, 2022-01-25:
We noticed that in the ENSENSO_INSTALL folder (e.g. `/opt/ensenso') are files to seem to link to certain memory addresses in the eeprom of the camera. It seems that without these files, the link between rgb and depth camera is not loaded correctly when launching the ros driver.

Am I correct in thinking that on startup the SDK loads the link calibration from the camera eeprom using the address in this file?

Does this mean that when launching the driver from another PC (with a cleanly installed SDK) with the same json file for the same camera, this link calibration is not loaded, since the address to the eeprom of the camera is different?

If so, this means that part of the camera configuration is 'hidden' in the file (as in: it's not enough to have the data is the eeprom and the json to complete reconstruct the setup). What would be the right workflow to fix this? Basically run the calibration routine on the new system again? Or is there some way to save/load this file somewhere and use it on a different installation?

saierd, 2022-01-25:
The files in `/opt/ensenso/cameras/` are used to emulate the EEPROM for devices which don't have a real EEPROM. You can simply copy those files to a different PC when you want to move the camera between different PCs. The file name includes the serial number of the device.

For devices that have an EEPROM all information is stored in the EEPROM, as expected.

This affects mostly IDS GigE Vision cameras with firmware < 2.4. Later firmware versions do support the real EEPROM and the Ensenso SDK also implements it since version 3.2 (see the [upgrade notes](https://www.ensenso.com/manual/3.2/about/notes.html#upgrade-notes)). If you want to use this already, you have to explicitly delete the EEPROM file and upgrade the firmware manually, since we still ship an older one.

The upcoming version 3.3 of the SDK will ship a newer firmware for IDS GigE Vision cameras and therefore support the EEPROM out of the box.

sepjansen, 2022-01-25:
Ah, that makes sense!

Good to hear this workaround is no longer needed with the new firmware and SDK.
We'll work around it for now and upgrade when SDK v3.3 is released.

closed 2022-01-25 by sepjansen