ensenso/ros_driver #73: A question about segmentation fault (core dumped)

shenjiyuan123, 2022-02-09:
I met the problem about segmentation fault when I run this `rosrun ensenso_camera ensenso_camera_node`
The result is like below:
`~/ensenso_ros1_ws$ rosrun ensenso_camera ensenso_camera_node
[ INFO] [1644395246.946467651]: Initializing nodelet with 12 worker threads.
Segmentation fault (core dumped)
`
Environment:
- ubuntu 20.04
- ensenso  s10
- sdk v3

saierd, 2022-02-09:
Do you run the latest commit 0e6822103b1f495105139d4a1e938546a2054159, which added support for S-series cameras?

If so, where exactly does it crash? Can you build a debug version of the driver and post the stack trace?

shenjiyuan123, 2022-02-20:
Thanks, I have found the reason. I can use roslaunch to start the topic instead.

closed 2022-02-21 by saierd