ensenso/ros_driver #74: Support for X36 series camera

juanmed, 2022-02-28:
Hi,

Thank you for this ros driver package. Does this repository support the X36-BL camera? I am evaluating a X36-BL camera and it is important for me to use it together with ROS. I saw pull request #72 and branch `xr-series` but it was not clear whether or not X36 is supported since the pull request is not merged. 

Also, the X36 has an RGB camera mounted on top and has been calibrated to output colored point clouds in Windows 10. Is this functionality also supported in this package, i.e., will this package load the already calibrated RGB-X36 setup and output colored point clouds? 

Thank you.

saierd, 2022-02-28:
Yes, it does support the X36. #72 will add support for the XR-series, which is different from the X-series (without R), but it is not quite finished yet.

The RGB camera is also supported via the [`texture_point_cloud` action](http://wiki.ros.org/ensenso_driver/Nodes/ensenso_camera_node#Provided_Actions). This uses the same NxLib functionality as NxView under the hood. See [this tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/TexturedPointCloud) for an example.

juanmed, 2022-02-28:
Thanks for your prompt reply. I will close this issue now.

closed 2022-02-28 by juanmed