ensenso/ros_driver #75: RGB frame registered depth image

juanmed, 2022-03-10:
Hi,

As mentioned in #74, I am evaluating an X36 camera. Our application requires the depth image registered to the RGB frame. According to #55, the ros_driver, although it outputs the colored point cloud, it does not output the depth image in the RGB frame. I have several questions:
a) Is this still the case? 
b) I tried the proposed solution in #55 (use depth_image_proc/register). However I found that the depth image registered is not correct. See below.

RGB image (it seems that ros_driver publishes rgb channels in inverted order)
[image]

Depth image (from the /stereo node)
[image]

Depth registered (obtained using  depth_image_proc/register as per suggested in #55).
[image]

I have used  depth_image_proc/register before, so I am pretty sure I am configuring it correctly. After checking the Tf for the transform between stereo and rgb camera, it seems to me that the translation is too large, but it matches the "Link" frame information stored inside the RGB camera eeprom... maybe something is weird with the units?

```
	"Link": {
		"Rotation": {
			"Angle": 0.0139332363601961714,
			"Axis": [
				0.62916486024742091,
				0.662676914584964138,
				0.406215319142441633
			]
		},
		"Target": "181961",
		"Translation": [
			-188.890016230833993,
			70.3128215477127725,
			-16.9718587224561368
		]
	},
```

[image]

You comments will be appreciated.

saierd, 2022-03-10:
Yes, this is still the case. We have the [point cloud texturing](http://wiki.ros.org/ensenso_driver/Tutorials/TexturedPointCloud), which produces a colored PCL point cloud, but no action projecting into the mono image. The external `depth_image_proc/register` node seems like a good ROS-style solution as long as we can get it to work.

The translation in the NxLib is in mm. When reading it, it should be converted to meters [here](https://github.com/ensenso/ros_driver/blob/42014bd0fa58ef4840a1e319c533a23958b0f9d1/ensenso_camera/src/pose_utilities.cpp#L80-L86), as expected by ROS. I read the code briefly and this looks correct so far. That function gets used to publish the TF transform, so it should definitely be different from the Link in the NxLib. Could you check that again, please?

juanmed, 2022-03-11:
Thanks for your answer. This is the info in TF:

```
transforms: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1646368513
        nsecs: 876365322
      frame_id: "181961_optical_frame"
    child_frame_id: "4103133522_optical_frame"
    transform: 
      translation: 
        x: 0.188321864392
        y: -0.071221193104
        z: 0.0193337047363
      rotation: 
        x: -0.00438311589862
        y: -0.00461657969713
        z: -0.0028299241361
        w: 0.999975733214
---

The translation is indeed in meters, and the signs are all opposite. This seems fine since it is expressing the transform from `181961_optical_frame`->`4103133522_optical_frame`, and the one in the .json file inside RGB camera's eeprom is the opposite, right? 
The quaternion is more difficult to verify.

saierd, 2022-03-11:
Yes, the conversion seems correct.

The origin of the stereo camera is in the calibrated optical projection center of the left sensor, so an offset in x direction is expected. 18cm seems quite high for this setup, but might be plausible depending on the exact viewing directions that were calibrated for the cameras.

Could you please check your link calibration in NxView by

* opening both devices in the calibration wizard and measuring the calibration and
* opening both devices in NxView and checking that the point cloud is colored correctly.

gavanderhoorn, 2022-03-11:
Off-topic, but those frame names seem to violate [naming rules](http://wiki.ros.org/Names#Valid_Names). Specifically: resource names cannot start with a digit.

juanmed, 2022-03-14:
Thanks for your answer. 
- I had the camera for 1 week only and I dont have it anymore. I recorded rosbags but cannot access the tools in NxView.
- I can confirm that the point cloud in NxView (and also using texture point cloud in ROS) was correctly colored. So it seems that, at least for NxLibrary everything is working correctly.

If its of help, here are the parameter files (json) for the stereo and the rgb cameras. This is were I got the Link transformation in the first comment (I had to change the extension to .docx for github to allow the upload, after downloading simply change the extension to .json):

[stereo_params.docx](https://github.com/ensenso/ros_driver/files/8243111/stereo_params.docx)

[rgb_params.docx](https://github.com/ensenso/ros_driver/files/8243112/rgb_params.docx)

saierd, 2022-03-14:
Thank you, we will try it out ourselves.

juanmed, 2022-03-14:
Thanks. Out of curiosity, I was checking the nxlib python interface. I have a bunch of rosbags with all the topics and messages that `mono_stereo_nodelets.launch` would produce (depth image/rgb/point clouds). Would it be possible to load and rectify the depth/rgb images in python, and output an RGBD image using the funtions in nxlib-python-interface (CM_RENDER_POINT_MAP, ITEM_TEXTURE = true)? (As in https://github.com/ensenso/nxlib-python-interface/issues/10)

juanmed, 2022-03-14:
> 

This is the way I put together my launch file to rectify and register the depth image: 

```xml
<launch>
  <arg name="serial_stereo" doc="Serial of the Ensenso stereo camera" />
  <arg name="serial_mono" doc="Serial of a mono camera supported by the NxLib" />
  <arg name="stereo_ns" doc="Namespace for the stereo camera" />
  <arg name="mono_ns" doc="Namespace for the mono camera" />
  <arg name="target_frame_name" value="Workspace" doc="The root the tf tree"/>

  <include file="$(find ensenso_camera)/launch/mono_stereo_nodelets.launch">
    <arg name="serial_stereo" value="$(arg serial_stereo)" />
    <arg name="serial_mono" value="$(arg serial_mono)" />
    <arg name="stereo_ns" value="$(arg stereo_ns)" />
    <arg name="mono_ns" value="$(arg mono_ns)" />
  </include>

  <node pkg="nodelet" type="nodelet" name="nodelet_manager" args="manager" />

  <node pkg="nodelet" type="nodelet" name="depth_register_nodelet"
        args="load depth_image_proc/register nodelet_manager" output="screen">
    <remap from="rgb/camera_info" to="/mono/rectified/camera_info"/>
    <remap from="depth/camera_info" to="/stereo/depth/camera_info"/>
    <remap from="depth/image_rect" to="/stereo/depth/image_rect"/>
    <param name="queue_size" type="int" value="30"/>
  </node>

  <node pkg="nodelet" type="nodelet" name="registered_cloud"
        args="load depth_image_proc/point_cloud_xyzrgb nodelet_manager" output="screen">
    <remap from="rgb/camera_info" to="/mono/rectified/camera_info"/>
    <remap from="rgb/image_rect_color" to="/mono/rectified/image"/>
    <remap from="depth_registered/image_rect" to="/depth_registered/image_rect"/>
  </node>

  <group ns="/stereo">
    <node pkg="image_proc" type="image_proc" name="depth_rectify" output="screen">
      <remap from="image_raw" to="/stereo/depth/image"/>
      <remap from="camera_info" to="/stereo/depth/camera_info"/>
      <remap from="image_rect" to="/stereo/depth/image_rect"/>
      <param name="interpolation" value="0" />
    </node>
  </group>

</launch>
```

saierd, 2022-03-14:
> Thanks. Out of curiosity, I was checking the nxlib python interface. I have a bunch of rosbags with all the topics and messages that `mono_stereo_nodelets.launch` would produce (depth image/rgb/point clouds). Would it be possible to load and rectify the depth/rgb images in python, and output an RGBD image using the funtions in nxlib-python-interface (CM_RENDER_POINT_MAP, ITEM_TEXTURE = true)? (As in [ensenso/nxlib-python-interface#10](https://github.com/ensenso/nxlib-python-interface/issues/10))

Images from rosbags cannot be used in the NxLib anymore. It requires a hardware device as a form of license.

If you saved images from NxView they are signed and you can use them as file cameras as described here: https://www.ensenso.com/manual/3.2/guides/debugging-and-simulation/file-cameras.html

juanmed, 2022-03-17:
I keep trying to solve this challenge. While looking at the raw point clouds saved using `request_data` script in `ensenso_camera` package, I notice that the "optical center" of the point clouds seems no to be the same as the root frame published in TF (in my case the frame of the stereo camera: 181961_optical_frame). This is visible because usually there is some point cloud noise that starts from the optical center of the stereo camera and extends all the way to the reconstructed scene. However, when visualizing the point clouds and TF in rviz, the "origin" of that noise is different from the optical frame. The following images show this: 
[image]

[image]

[image]

I might be wrong but something does not feel good about this. 

Also when I recorded my rosbags, I also took note of the approximate distance from the camera to an specific point in the scene. When I measure in the point cloud from 181961_optical_frame to the same point, it does not match. But it matches when I measure from the "optical center" of the point (where the noise starts). 

Any comments are welcome.

benthie, 2022-03-24:
Thank you for providing the information and configuration of your setup. We had a look into your issue and compiled a detailed [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/RegisteredDepthImage) on to how use the [`depth_image_proc/register`](http://wiki.ros.org/depth_image_proc) node to register a depth image from an Ensenso stereo camera to the RGB frame of a linked RGB mono camera.

In addition, we fixed a few minor bugs and further improved the our ROS package. See [this](https://github.com/ensenso/ros_driver/pull/81) merge request.

Regarding your setup there are a two things I noticed:
- The first thing is that the two Ensenso X-Camera setups you provided photos of have different baselines: The [first one](https://user-images.githubusercontent.com/21977733/157796431-161d8745-a0f5-408a-ac1b-fbcce97a3d3d.png) is a 200mm baseline setup whereas the [second one](https://user-images.githubusercontent.com/21977733/158152517-6174c25e-a49b-48b7-bd60-a24637fd6629.png) has a 400mm baseline. In case you somehow forgot to re-calibrate the 200mm setup it would explain the `x` translation of 18cm.
- The other thing is the point cloud which is at the same position as the camera's optical frame (depicted in your last post in the [second image](https://user-images.githubusercontent.com/21977733/158823259-0eef2c9a-bcea-4d30-b305-57db10c2c57c.png)): If your camera is workspace calibrated and you do not specify a target frame, the Ensenso camera node does not publish the transform from camera to workspace and thus `tf` cannot transform the point cloud to the correct position. From the JSON configuration files you provided I can see that your stereo camera is workspace calibrated. You could verify that you started the node with a target frame. If not, you should see [this warning](https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/src/stereo_camera.cpp#L457) in yours logs.

I will close this issue for now. If you have further questions feel free to reopen.

closed 2022-03-24 by benthie

juanmed, 2022-03-27:
Hi @benthie 

I appreciate the effort and detail you put into compiling the tutorial above and answers to my issue. I would like to go through each point. After reading your comments and the tutorial above, I realize that I probably recorded the data with the incorrect target_frame setting. Anyhow, my objective now is to be able to use the rosbags I stored to successfully register the depth image `stereo/depth/image` to the mono camera frame. 

1. The first image I provided was only for reference. The only camera I ever used was exactly as in the second image. This source of problem can be eliminated.
2. It seems you are right on this one: I did not specified any target_frame when I launched `mono_stereo_nodelets.launch`. I also remember I saw the warning you mentioned.  I remember I did this after assuming that launching the nodelets this way would give me all data from each camera without any transformation (each expressed on its own optical center) and later I could transform it however I wanted.

Since I recorded my data before the fixes and improvements mentioned above, the following questions refers to this repo before those fixes:
1. If a `target_frame` is specified, are all `stereo/point_cloud`,  and even `stereo/depth/camera_info` and `stereo/depth/image`  represented in the specified `target_frame`?   Being this my first experience with Ensenso cameras, I originally assumed only the point_cloud is transformed and the depth image (and camera_info) would always be represented in the optical frame of the stereo camera disregard of whatever `target_frame` contains.
2. I am confounded by the information in the `header` of the `stereo/point_cloud`, `stereo/depth/camera_info` and `stereo/depth/image`. They all contain "frame_id:181961_optical_frame" but it seems this data is all expressed in other frame, i.e. Workspace. 
3. I can also find the `Workspace` Link in the stereo_params.json parameter file of the stereo camera. I assume you refer to lines 490-505 of that file, right? 
4. You mentioned the following:
> If your camera is workspace calibrated and you do not specify a target frame, the Ensenso camera node does not publish the transform from camera to workspace and thus tf cannot transform the point cloud to the correct position

So, how can I use the transformations available in the .json and tf (stored in rosbags) to successfully register the depth image into the rgb camera frame?  I dont have the camera anymore with me so all I am left to evaluate if the camera is appropriate for our application are the rosbags. 
 
We are evaluating ensenso's x36 camera for a commercial application, so your continued support is much appreciated. 

Thank you

juanmed, 2022-04-04:
@benthie I can share the .json files with you and the rosbags. If you do not need the rosbags, please let me know exactly which images you need (and possible camera intrinsic parameters??).  Please send me an email to medrano(at)g(dot)skku(dot)edu. Thanks.