ensenso/ros_driver #89: Failed to load nodelet /mono/Ensenso_

dilerbatu, 2022-07-19:
Hi Everyone, 
I have ensenso S series camera and I already installed nxView and I can see my view. I am trying to connect with ROS. I downloaded the repository and built it. When I run 

roslaunch ensenso_camera register_depth_image.launch it gives this error:

[image]

Thank you !

saierd, 2022-07-27:
It seems like something crashed with a segmentation fault, but from the error message it is not clear, whether it was the nodelet manager, one of the camera nodes or the NxLib.

- Do you have the [version 1.7](https://github.com/ensenso/ros_driver/releases/tag/v1.7.0) or later, which added support for the S-series in the ROS interface?
- Do the camera nodes run individually if you simply rosrun `ensenso_camera_node`?
- Can you post the content of the log files, which are mentioned in the error message?

saierd, 2023-09-25:
I will close this for now. Feel free to reopen if the issue persists and you have further information.

closed 2023-09-25 by saierd