ensenso/ros_driver #91: Enable CUDA

diogo-sr, 2022-10-14:
Hi everyone,

I am trying to enable the usage of CUDA when starting the cameras but I am not finding a way of doing that. According to the documentation [here](https://www.ensenso.com/manual/3.0/tree/cuda/enabled.html#cuda-enabled) it is a parameter from the cuda node, which I cannot find a way of accessing using this wrapper. 

Can someone point me in the correct direction, please?

saierd, 2022-10-14:
Yes, there is currently no specific parameter for enabling CUDA.

As a workaround you can write to the `/CUDA/Enabled` node directly with the [`access_tree`](http://docs.ros.org/en/api/ensenso_camera_msgs/html/action/AccessTree.html) action.

diogo-sr, 2022-10-14:
Thank you for the reply, I will try that. Is it also an option to make a PR to expose this parameter in the RequestData action? Because some commands accept CUDA as a parameter that overrides the global settings

saierd, 2022-10-14:
Yes, that would be a good feature. :+1: 

We could also let the user set the default parameter as a node parameter. But unfortunately all of the cameras in the same process share the `/CUDA/Enabled` node so that might be a but confusing.

diogo-sr, 2022-10-14:
With the `access_tree` option the change does not seem to be persistent. I call:

```
rostopic pub /camera_left/access_tree/goal ensenso_camera_msgs/AccessTreeActionGoal "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs: 0
  id: ''
goal: {parameter_set: '', path: '/Cuda/Enabled', erase: false, set_null: false, json_value: 'true'}" 
publishing and latching message. Press ctrl-C to terminate
```

and the call result succeeds with the json value supposedly being updated, but if I then ask for the tree status after setting the value I get:]

```
result: 
  exists: True
  json_value: "{\"CUDA\":{\"Available\":true,\"Device\":0,\"Devices\":[{\"ClockRate\":1882,\"ComputeCapability\"\
  :\"8.6\",\"Cores\":0,\"Integrated\":false,\"Memory\":12053,\"Name\":\"NVIDIA GeForce\
  \ RTX 3060\"}],\"Enabled\":false,\"ErrorText\":\"\",\"StaticBuffers\":2,\"UseFloat16\"\
  :false,\"Version\":11070},
```

diogo-sr, 2022-10-14:
It does work, I had a typo  as it indeed must be `/CUDA/Enabled`