ensenso/ros_driver #105: ensenso_camera_node exit code -9

erblinium, 2023-02-01:


erblinium, 2023-02-01:
Is looks like it is a memory problem. The memory of the `ensenso_camera_node` always increases until it crashes with the error code. The virtual camera also disappears from NxView after the crash, and I have to restart machine to get it back.

erblinium, 2023-02-01:
Investigating further I see that I have another node running of type `image_stream` in the `ensenso_camera` package. This node's CPU usage increases as the nodes run.

erblinium, 2023-02-02:
Any ideas what might cause this @benthie? I believe it has to do with the requested data from the `image_stream` node...

benthie, 2023-02-02:
I will have a look at it.
- Are you using ROS1 or ROS2?
- Are you using the virtual camera directly or as a file camera?

Let me summarize what you are doing: you have an `ensenso_camera_node` running and besides that only the `image_stream` script?

erblinium, 2023-02-02:
> I will have a look at it.
> 
>     * Are you using ROS1 or ROS2?
ROS2
> 
>     * Are you using the virtual camera directly or as a file camera?
Virtual camera directly
> 
> 
> Let me summarize what you are doing: you have an `ensenso_camera_node` running and besides that only the `image_stream` script?
Yes, I start them both from a launch file. I also have some other nodes running TF, robot state publishers, but I do not think they cause any problems here.

benthie, 2023-02-02:
Thank you for the details. I will get back to you if I have found something .

benthie, 2023-02-02:
Which Ensenso SDK version do you use?

erblinium, 2023-02-02:
> Which Ensenso SDK version do you use?

3.3.1417

benthie, 2023-02-03:
I think I found a solution for the high CPU load of the python scripts. The fix is to use a `MultiThreadedExecutor` in combination with a `ReentrantCallbackGroup`. I am currently testing it on my machine and it looks good with regard to the CPU usage. For the memory leak I created a separate [issue](#107). I will keep you posted as soon as I have a working fix that also passes all tests.

closed 2023-02-10 by benthie

benthie, 2023-02-10:
FYI, the fixes for both this issue and https://github.com/ensenso/ros_driver/issues/107 are part of the currently released [v2.1.0](https://github.com/ensenso/ros_driver/releases/tag/v2.1.0).

PR #108 (2023-02-10): Fix ros2 bugs
Fixes #105
Fixes #107