ensenso/ros_driver #114: Build error. Jetson Nano 4G with Melodic. Jetpack 4.6

onthesofa, 2023-09-01:
Installed ensenso-sdk-3.5.1394-arm64.deb which works (at about 0.6 fps) then cloned the ros_driver and ran catkin_make. This error:

...catkin_ws_ensenso/src/ros_driver/ensenso_camera/include/ensenso_camera/ros2/ensenso_msgs/image_point.h:8:10: fatal error: ensenso_camera_msgs/ImagePoint.h: No such file or directory
 #include "ensenso_camera_msgs/ImagePoint.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Sure enough ImagePoint.h doesn't exist anywhere on the system.

benthie, 2023-09-06:
Hi @onthesofa, 

since you only mentioned that you cloned the `ros_driver` and ran `catkin_make`, I can't tell if you are an experienced ROS user or new to ROS, so let me ask: have you
- tested your ROS Melodic setup with other packages, are they working?
- installed the ROS dependencies for our `ros_driver`?

We stopped supporting Melodic a few months ago with the EOL of Ubuntu 18.04. At that point the latest Ensenso SDK we tested with Melodic was 3.4.743. You could try this version as an alternative.

onthesofa, 2023-09-06:
Hi @benthie

It's a good question. I have a fair amount or experience with ROS (mostly traumatic!). I have other packages installed and running. During the first attempts to install ros_driver there were many package dependencies not found, but one-by-one I installed them and now we are at the compilation stage. You can see from the error I posted that the error comes from 'ensenso_camera/ros2/ensenso_msgs/image_point.h' being unable to include  "ensenso_camera_msgs/ImagePoint.h". I think the package contains both ROS and ROS2 together and something is confused.

i.e.

#ifdef ROS2
#include "ensenso_camera_msgs/msg/image_point.hpp"

onthesofa, 2023-09-06:
I just saw your previous post - I will try that. Thanks.

onthesofa, 2023-09-06:
Hi @benthie 

This is the result:

harper@nano-4G-4:~/cws$ catkin build
--------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/harper/catkin_ws/devel:/opt/ros/melodic
Workspace:                   /home/harper/cws
--------------------------------------------------------------------------
Build Space:        [exists] /home/harper/cws/build
Devel Space:        [exists] /home/harper/cws/devel
Install Space:      [unused] /home/harper/cws/install
Log Space:         [missing] /home/harper/cws/logs
Source Space:       [exists] /home/harper/cws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 8.1 seconds ]             
Starting  >>> ensenso_camera_msgs                                              
Starting  >>> ensenso_description                                              
Finished  <<< ensenso_description                  [ 9.0 seconds ]             
Finished  <<< ensenso_camera_msgs                  [ 1 minute and 41.9 seconds ]
Starting  >>> ensenso_camera                                                   
_______________________________________________________________________________
Errors     << ensenso_camera:make /home/harper/cws/logs/ensenso_camera/build.make.000.log
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/harper/cws/devel/.private/ensenso_camera/lib/ensenso_camera/ensenso_camera_mono_node'. Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/harper/cws/devel/.private/ensenso_camera/lib/ensenso_camera/texture_point_cloud'. Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/harper/cws/devel/.private/ensenso_camera/lib/ensenso_camera/ensenso_camera_node'. Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ensenso_camera_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ensenso_camera_mono_node.dir/all] Error 2
make[1]: *** [CMakeFiles/texture_point_cloud.dir/all] Error 2
make[2]: *** No rule to make target '/usr/lib/aarch64-linux-gnu/libopencv_objdetect.so.3.2.0', needed by '/home/harper/cws/devel/.private/ensenso_camera/lib/libensenso_camera_nodelet.so'. Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ensenso_camera_nodelet.dir/all] Error 2
make: *** [all] Error 2
cd /home/harper/cws/build/ensenso_camera; catkin build --get-env ensenso_camera | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << ensenso_camera:make                  [ Exited with code 2 ]      
Failed    <<< ensenso_camera                       [ 3 minutes and 59.3 seconds ]
Abandoned <<< ensenso_camera_test                  [ Unrelated job failed ]    
[build] Summary: 3 of 5 packages succeeded.                                    
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 5 minutes and 50.2 seconds total.                             
[build] Note: Workspace packages have changed, please re-source setup files to use them.
harper@nano-4G-4:~/cws$

benthie, 2023-09-06:
Okay. It looks like now you are just missing the OpenCV package. Please try installing it with:

```
sudo apt-get -y install libopencv-dev python3-opencv
```

benthie, 2023-09-06:
And if you are then still missing some transformation packages, please try installing them with:

```
sudo apt-get -y install ros-${ROS_DISTRO}-tf-transformations
sudo pip3 install transforms3d
```

onthesofa, 2023-09-06:
Thank @benthie

The OpenCV  install was enough.

harper@nano-4G-4:~/cws$ catkin build
--------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/harper/catkin_ws/devel:/opt/ros/melodic
Workspace:                   /home/harper/cws
--------------------------------------------------------------------------
Build Space:        [exists] /home/harper/cws/build
Devel Space:        [exists] /home/harper/cws/devel
Install Space:      [unused] /home/harper/cws/install
Log Space:          [exists] /home/harper/cws/logs
Source Space:       [exists] /home/harper/cws/src
DESTDIR:            [unused] None
--------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------------
[build] Found '5' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> ensenso_camera_msgs                                              
Starting  >>> ensenso_description                                              
Finished  <<< ensenso_description                [ 0.7 seconds ]               
Finished  <<< ensenso_camera_msgs                [ 10.5 seconds ]              
Starting  >>> ensenso_camera                                                   
Finished  <<< ensenso_camera                     [ 16.1 seconds ]              
Starting  >>> ensenso_camera_test                                              
Finished  <<< ensenso_camera_test                [ 11.5 seconds ]              
[build] Summary: All 4 packages succeeded!                                     
[build]   Ignored:   1 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 38.5 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

benthie, 2023-09-06:
I am glad to hear that. We will update our installation documentation accordingly ;)

closed 2023-09-06 by benthie