ensenso/ros_driver #116: ensenso_camera_node runs (with a warning), but  rosrun ensenso_camera request_data results in "Illegal instruction (core dumped)"

onthesofa, 2023-09-08:
Jetson Nano 4G Jetpack 4.6. ROS Melodic. SDK ensenso-sdk-3.4.743-arm64.deb. NxView works.

Trying to follow this tutorial: https://wiki.ros.org/ensenso_driver/Tutorials/FirstSteps

Level beginner..

harper@nano-4G-4:~$ rosrun ensenso_camera ensenso_camera_node
[ INFO] [1694175730.456104834]: Initializing nodelet with 4 worker threads.
[ INFO] [1694175739.225077617]: Opened camera with serial number 'redacted'.
[ WARN] [1694175739.225393977]: Camera 'redacted' has an internal link (i.e. it is either extrinsically calibrated (workspace- or hand-eye) or has a link to another camera), but camera and target frame are equal, which means that neither a link nor a target frame has been provided. The images and 3d data retreived from the camera are transformed by the NxLib with the transform stored in the camera's link node, however, this transform is not known to tf. Please provide a link or target frame in order for the transform to be published.

Requesting data from the camera fails:

harper@nano-4G-4:~$ rosrun ensenso_camera request_data
Illegal instruction (core dumped)

Not sure where to look for clues.

benthie, 2023-09-08:
After a quick research I think 'Illegal instruction (core dumped)' could have something to do with your memory, maybe not having enough of it. You could try to request less data from the camera and see if that works. By default, the script requests all possible data, see https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/scripts/request_data.

saierd, 2023-09-08:
Just to make sure: the command that crashes with the `Illegal instruction` message is the `request_data` one? The `ensenso_camera_node` continues to run in the background?

If so, there is probably some problem with your installation of ROS, Python or some other library that is used by those two. The `request_data` node is a [Python script](https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/scripts/request_data) that only uses ROS functionality to query the data. The only code from this repository and the Ensenso SDK that runs in that process is that Python script, so the crash must be caused by something on a lower level.

Unfortunately I have no idea how to debug that properly, so let's confirm that the problem is with the `request_data` script first.

saierd, 2023-09-08:
I found this issue that might very well be it: https://github.com/numpy/numpy/issues/18131

Could you try to set the environment variable `OPENBLAS_CORETYPE=ARMV8` as suggested in that issue?

onthesofa, 2023-09-08:
I tried other scripts already. e.g.

harper@nano-4G-4:~$ rosrun ensenso_camera request_data_mono
Illegal instruction (core dumped)

I also just tried lightening the load in request data to just raw images (not sure if tis is what you meant):

    goal.request_raw_images = ros2py.get_param(node, "raw_images", True)
    #goal.request_rectified_images = ros2py.get_param(node, "rectified_images", True)
    #goal.request_disparity_map = ros2py.get_param(node, "disparity_map", True)
    #goal.request_depth_image = ros2py.get_param(node, "depth_image", True)
    #goal.request_point_cloud = ros2py.get_param(node, "point_cloud", True)
    #goal.request_normals = ros2py.get_param(node, "normals", True)

The camera node seems to be OK, with the following topics created (but idle?):

harper@nano-4G-4:~$ rostopic list
/access_tree/cancel
/access_tree/feedback
/access_tree/goal
/access_tree/result
/access_tree/status
/calibrate_hand_eye/cancel
/calibrate_hand_eye/feedback
/calibrate_hand_eye/goal
/calibrate_hand_eye/result
/calibrate_hand_eye/status
/calibrate_workspace/cancel
/calibrate_workspace/feedback
/calibrate_workspace/goal
/calibrate_workspace/result
/calibrate_workspace/status
/camera_info
/depth/camera_info
/depth/image
/depth/projected_depth_map
/diagnostics
/disparity_map
/execute_command/cancel
/execute_command/feedback
/execute_command/goal
/execute_command/result
/execute_command/status
/fit_primitive/cancel
/fit_primitive/feedback
/fit_primitive/goal
/fit_primitive/result
/fit_primitive/status
/get_parameter/cancel
/get_parameter/feedback
/get_parameter/goal
/get_parameter/result
/get_parameter/status
/locate_pattern/cancel
/locate_pattern/feedback
/locate_pattern/goal
/locate_pattern/result
/locate_pattern/status
/point_cloud
/point_cloud_color
/project_pattern/cancel
/project_pattern/feedback
/project_pattern/goal
/project_pattern/result
/project_pattern/status
/project_telecentric/cancel
/project_telecentric/feedback
/project_telecentric/goal
/project_telecentric/result
/project_telecentric/status
/projected_point_cloud
/raw/left/camera_info
/raw/left/image
/raw/right/camera_info
/raw/right/image
/rectified/left/camera_info
/rectified/left/image
/rectified/right/camera_info
/rectified/right/image
/request_data/cancel
/request_data/feedback
/request_data/goal
/request_data/result
/request_data/status
/rosout
/rosout_agg
/set_parameter/cancel
/set_parameter/feedback
/set_parameter/goal
/set_parameter/result
/set_parameter/status
/texture_point_cloud/cancel
/texture_point_cloud/feedback
/texture_point_cloud/goal
/texture_point_cloud/result
/texture_point_cloud/status
/tf
/tf_static

onthesofa, 2023-09-08:
That cleared the error. Thank you. Because I have Melodic which is python2 I had to change the shebang and it appears to be working! Back to the tutorial :)

harper@nano-4G-4:~$ rosrun ensenso_camera request_data
[INFO] [1694183126.632752]: Connecting to action server request_data ...
[INFO] [1694183126.851811]: Connected!
[INFO] [1694183140.185542]: Data successfully requested.
[INFO] [1694183145.596383]: Data successfully requested.

saierd, 2023-09-08:
Nice :+1: You can probably fix it more permanently by installing a different version of numpy. Or just keep the variable set :wink:

closed 2023-09-08 by saierd