ensenso/ros_driver #117: CMake cant find Ensenso SDK

belalhmedan90, 2023-11-29:
Hi, 
I have installed Ensenso SDK on Ubuntu 20.04; apt list shows: `ensenso-sdk/now 3.5.1419 amd64 [installed,local]`
when I try to build the catkin workspace I get the following error: 
```
CMake Error at CMakeLists.txt:68 (message):
  Please install the Ensenso SDK

Failed     << ensenso_camera:cmake 
```
can you please tell me how can I resolve this issue? thanks in advance.

benthie, 2023-11-29:
Did you set the `ENSENSO_INSTALL` environment variable? It is described [here](https://www.ensenso.com/manual/3.5/about/installation.html) under **Storage of Settings**.

belalhmedan90, 2023-11-29:
@benthie thanks, `export ENSENSO_INSTALL=/opt/ensenso` solved the problem in my case.

closed 2023-11-29 by belalhmedan90