ensenso/ros_driver #118: NxLibException 17 when running the camera

belalhmedan90, 2023-11-29:
Hi,

When I run: `roslaunch ensenso_camera register_depth_image.launch` I get an exception.
Im using ROS1, ubuntu `20.04`, Camera type: `C75-8-M`
 
```
process[DIP_Register-6]: started with pid [58579]
[ INFO] [2023-11-29 14:56:34]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2023-11-29 14:56:34]: Loading nodelet /mono/Ensenso_ of type ensenso_camera/mono_camera_node to manager /manager_ with the following remappings:
[ INFO] [2023-11-29 14:56:34]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2023-11-29 14:56:34]: Initializing nodelet with 12 worker threads.
[ INFO] [2023-11-29 14:56:34]: Loading nodelet /DIP_Register of type depth_image_proc/register to manager /manager_ with the following remappings:
[ INFO] [2023-11-29 14:56:34]: /depth/camera_info -> /stereo/depth/camera_info
[ INFO] [2023-11-29 14:56:34]: /depth/image_rect -> /stereo/depth/image
[ INFO] [2023-11-29 14:56:34]: /depth_registered/camera_info -> /mono/depth/camera_info
[ INFO] [2023-11-29 14:56:34]: /depth_registered/image_rect -> /mono/depth/image
[ INFO] [2023-11-29 14:56:34]: /rgb/camera_info -> /mono/rectified/camera_info
[ INFO] [2023-11-29 14:56:34]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2023-11-29 14:56:34]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2023-11-29 14:56:36]: Opened camera with serial number '232155'.
[ INFO] [2023-11-29 14:56:37]: Opened camera with serial number '232155-Color'.
[ERROR] [2023-11-29 14:56:38]: NxLibException 17 (ExecutionFailed) for item /Execute/232155-Color
[ERROR] [2023-11-29 14:56:38]: CaptureTimeout: Image capture timed out.
```

Can you please tell me how can I solve that please? thanks.

benthie, 2023-11-29:
Unfortunately, we haven't had the time yet to test C-Series cameras with ROS.

The C-Series has a [special trigger mode](https://www.ensenso.com/manual/3.5/guides/camera-operation/c.html#device-internal-trigger). Maybe using the internal trigger mode will solve your problem.

benthie, 2023-11-30:
I think that the capture timeout is specific to the C-Series camera. Issue #53 was about uEye based cameras and the uEye driver, while the C-Series is GigE Vision based and does not require an additional driver.

- Which version of the Ensenso SDK have you installed?
- Do you also see the capture timeouts when operating the camera from within NxView?
- If so, could you please record a log file?
  - Start recording by clicking: NxView -> NxTreeEdit button in the lower left corner -> "Debug" tab -> Start
  - Open your camera and operate it as you would with ROS and wait until the timeout occurs
  - Stop recording by clicking the previous button, which now says "Stop"
- Please contact the [IDS support](https://en.ids-imaging.com/support-ticket.html) and provide them with the recorded log file, your used SDK version and a link to this issue

benthie, 2023-12-11:
> Unfortunately, we haven't had the time yet to test C-Series cameras with ROS.

Now we had the time :tada:. There is a new release [v2.1.1](https://github.com/ensenso/ros_driver/releases/tag/v2.1.1) and a new [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera).

belalhmedan90, 2023-12-11:
@benthie  Thanks for your kind feedback, `roslaunch ensenso_camera mono_stereo_nodelets.launch stereo_serial:="232155" mono_serial:="232155-Color" node_name_mono:="232155_Color" link_frame:="optical_frame_232155"` is running well now, except I don't receive any data images/pointclouds. nxView gives an error: `CaptureTimeout: Waiting for images timed out`.

benthie, 2023-12-11:
Hi @belalhmedan90,

I just contacted the IDS support. They will get in touch with you soon. 

Until then you can:
- Check out our [first steps guide](https://www.ensenso.com/manual/3.5/about/first-steps.html) if you haven't read it yet
- Try connecting your camera directly to your computer
- Increasing your network MTU as described at the end of [this article](https://www.ensenso.com/manual/3.5/about/network-wizard.html)

I hope you soon find a solution for your problem.

belalhmedan90, 2023-12-20:
Hi again, 
I'm now able to see data using the SDK, thanks to the POE++ injector, 
however, I cant see any output using the past ros launch file (ROS1 Ubuntu 20.04), 
can you please tell me how can I `request_data` from the camera to see the output using the launch file, thanks in advance.

benthie, 2023-12-20:
Have you tried `rosrun ensenso_camera request_data`?

belalhmedan90, 2023-12-20:
> Have you tried `rosrun ensenso_camera request_data`?

yes I did, it worked with the [node](http://wiki.ros.org/ensenso_driver/Tutorials/FirstSteps), but not with the [launch file](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera)!

benthie, 2023-12-20:
- What does `rostopic list` show?
- Does it make a difference if you deactivate your Python `venv` before calling the request data script?
- Does it make a difference if you also provide a name for the stereo node, currently it is "Ensenso_", which might not be a valid name, but that's just a guess.

belalhmedan90, 2023-12-20:
1. ROS topic list
```sh
$ rostopic list
/diagnostics
/manager_/bond
/mono/access_tree/cancel
/mono/access_tree/feedback
/mono/access_tree/goal
/mono/access_tree/result
/mono/access_tree/status
/mono/execute_command/cancel
/mono/execute_command/feedback
/mono/execute_command/goal
/mono/execute_command/result
/mono/execute_command/status
/mono/get_parameter/cancel
/mono/get_parameter/feedback
/mono/get_parameter/goal
/mono/get_parameter/result
/mono/get_parameter/status
/mono/locate_pattern/cancel
/mono/locate_pattern/feedback
/mono/locate_pattern/goal
/mono/locate_pattern/result
/mono/locate_pattern/status
/mono/raw/camera_info
/mono/raw/image
/mono/raw/image/compressed
/mono/raw/image/compressed/parameter_descriptions
/mono/raw/image/compressed/parameter_updates
/mono/raw/image/compressedDepth
/mono/raw/image/compressedDepth/parameter_descriptions
/mono/raw/image/compressedDepth/parameter_updates
/mono/raw/image/theora
/mono/raw/image/theora/parameter_descriptions
/mono/raw/image/theora/parameter_updates
/mono/rectified/camera_info
/mono/rectified/image
/mono/rectified/image/compressed
/mono/rectified/image/compressed/parameter_descriptions
/mono/rectified/image/compressed/parameter_updates
/mono/rectified/image/compressedDepth
/mono/rectified/image/compressedDepth/parameter_descriptions
/mono/rectified/image/compressedDepth/parameter_updates
/mono/rectified/image/theora
/mono/rectified/image/theora/parameter_descriptions
/mono/rectified/image/theora/parameter_updates
/mono/request_data/cancel
/mono/request_data/feedback
/mono/request_data/goal
/mono/request_data/result
/mono/request_data/status
/mono/set_parameter/cancel
/mono/set_parameter/feedback
/mono/set_parameter/goal
/mono/set_parameter/result
/mono/set_parameter/status
/request_data/cancel
/request_data/feedback
/request_data/goal
/request_data/result
/request_data/status
/rosout
/rosout_agg
/stereo/access_tree/cancel
/stereo/access_tree/feedback
/stereo/access_tree/goal
/stereo/access_tree/result
/stereo/access_tree/status
/stereo/calibrate_hand_eye/cancel
[... 114 lines cut ...]
/stereo/rectified/right/image/compressedDepth/parameter_updates
/stereo/rectified/right/image/theora
/stereo/rectified/right/image/theora/parameter_descriptions
/stereo/rectified/right/image/theora/parameter_updates
/stereo/request_data/cancel
/stereo/request_data/feedback
/stereo/request_data/goal
/stereo/request_data/result
/stereo/request_data/status
/stereo/set_parameter/cancel
/stereo/set_parameter/feedback
/stereo/set_parameter/goal
/stereo/set_parameter/result
/stereo/set_parameter/status
/stereo/texture_point_cloud/cancel
/stereo/texture_point_cloud/feedback
/stereo/texture_point_cloud/goal
/stereo/texture_point_cloud/result
/stereo/texture_point_cloud/status
/tf
/tf_static
```

2. `venv` has no effect if deactivated.
3. I added a name to the stereo name, it didn't solve the problem.

benthie, 2023-12-20:
Might it be related to the Python version as in this [issue](https://github.com/ensenso/ros_driver/issues/116)?

benthie, 2023-12-20:
I am sorry, it has probably nothing to do with the Python version.

The launch file puts the two nodes into namespaces (`mono` and `stereo`). You have to provide the namespace to the `request_data` script. Have a look at the `color_point_cloud` script, there you can see the usage of the namespace parameter.

belalhmedan90, 2023-12-20:
Can you please tell me how can I run the `request_data` node in the correct namespace from the command-line? otherwise would you please provide a valid launch file to output a pointcloud to RVIZ? thanks.

benthie, 2023-12-20:
If you want to receive a point cloud in RVIZ, can't you just follow the steps in this [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera), which you referred to above? There the `color_point_cloud` script is used, which takes care of providing the default namespace parameters.

In RVIZ you then have to change the default frame from "map" to "optical_frame_232155" and should be able to see the data. At least for me this worked when I created the tutorial, both on ROS1 and ROS2.

belalhmedan90, 2023-12-20:
> If you want to receive a point cloud in RVIZ, can't you just follow the steps in this [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera), which you referred to above? There the [...]

That tutorial didn't work in my case, as the node for triggering the camera is not included in that launch file! so if I launch that launch file I don't get any output!

benthie, 2023-12-20:
Okay. Maybe I didn't explain that well enough in the tutorial. The node, that triggers the camera / captures images and then initiates the pipeline of commands that finally compute the desired point cloud, is indeed not included in the launch file. However, Step 2 in the tutorial says, that we have to run the `color_point_cloud` script, which:

- maps the namespaces and then
- triggers / captures on the stereo camera by sending the `RequestData` action and also
- triggers / captures on the mono camera by sending the `RequestDataMono` action

So instead of running the `request_data` script in a parallel terminal you should receive a colored point cloud by running the `color_point_cloud` script.

Let me know if you still do not get any data from the camera.

belalhmedan90, 2023-12-20:
I'm getting this error while running the color_point_cloud (either as ros node, or as python script, in both cases):
```sh
$ rosrun ensenso_camera color_point_cloud
Traceback (most recent call last):
  File "/home/belalhmedan/ids_ws/devel/lib/ensenso_camera/color_point_cloud", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/belalhmedan/ids_ws/src/ros_driver/ensenso_camera/scripts/color_point_cloud", line 78, in <module>
    main()
  File "/home/belalhmedan/ids_ws/src/ros_driver/ensenso_camera/scripts/color_point_cloud", line 74, in main
    ros2py.wrap_main_function(_main, "color_point_cloud")
  File "/home/belalhmedan/ids_ws/src/ros_driver/ensenso_camera/src/ensenso_camera/ros2.py", line 304, in wrap_main_function
    main(node_name)
  File "/home/belalhmedan/ids_ws/src/ros_driver/ensenso_camera/scripts/color_point_cloud", line 15, in _main
    mono_serial = ros2py.get_param(node, "mono_serial")
  File "/home/belalhmedan/ids_ws/src/ros_driver/ensenso_camera/src/ensenso_camera/ros2.py", line 321, in get_param
    return rospy.get_param("~" + name, default)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/client.py", line 467, in get_param
    return _param_server[param_name] #MasterProxy does all the magic for us
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/msproxy.py", line 123, in __getitem__
    raise KeyError(key)
KeyError: '~mono_serial'
```

benthie, 2023-12-20:
Try `rosrun ensenso_camera color_point_cloud _mono_serial:=232155-Color`.

belalhmedan90, 2023-12-20:
> rosrun ensenso_camera color_point_cloud _mono_serial:=232155-Color

It crashed:
[image]

benthie, 2023-12-20:
Did you make any specific settings in NxView that were necessary to get data? Or was the POE++-Injector the solution to your initial timeout problem? At any rate, could you export your settings from NxView via the settings dialog for both cameras and then use the saved `.json` files with ROS by passing them to the launch file?

```
roslaunch ensenso_camera \
    mono_stereo_nodelets.launch \
    stereo_serial:="232155" \
    mono_serial:="232155-Color" \
    node_name_mono:="232155_Color" \
    link_frame:="optical_frame_232155" \
    settings_stereo:=<path_to_stereo_settings> \
    settings_mono:=<path_to_mono_settings> 
```

belalhmedan90, 2023-12-21:
> Did you make any specific settings in NxView that were necessary to get data? Or was the POE++-Injector the solution to your initial timeout problem? At any rate, could you export your settings from [...]

POE++ injector was the solution, no changes to nxView were made, exporting settings json files and using them in launching the nodelet didn't work also.

benthie, 2023-12-21:
I am sorry to hear that this also doesn't solve your problem.

In order to further investigate your problem, could you please record the following logs and provide them to us:

- `.nxlog` file as described [above](https://github.com/ensenso/ros_driver/issues/118#issuecomment-1833384064). For this you need to open a TCP port via the launch file on the NxLib contained within the ROS node as described in this [issue](https://github.com/ensenso/ros_driver/issues/121#issuecomment-1845068023)
- The output of both the shell running the launch file and the shell running the script
- The complete NxLib tree as JSON while both cameras are opened. For this you can right-click on the root ("/") item in NxTreeEdit and then on "Copy value as JSON string".

Let me know if you have trouble recording the logs. And thank you for your patience.

belalhmedan90, 2023-12-21:
Here is the [link to the debug files](https://drive.google.com/file/d/1VFdiO6pH8ckewqlXaZPQrljjKRCWoBsP/view?usp=sharing)

benthie, 2023-12-21:
Could you please re-record the `.nxlog` file with an open TCP port on the ROS node (e.g. 25000) like this:

```
roslaunch ensenso_camera \
     mono_stereo_nodelets.launch \
     stereo_serial:="232155" \
     mono_serial:="232155-Color" \
     node_name_mono:="232155_Color" \
     link_frame:="optical_frame_232155" \
     settings_stereo:="../ensenso_stereo.json" \
     settings_mono:="../ensenso_mono.json" \
     tcp_port:=25000
```

Then open NxTreeEdit and connect to the corresponding instance with the above port number and repeat the recording.

benthie, 2023-12-21:
I also noticed that your MaximumTransmissionUnit (MTU) is set to 1500. Could you increase it to 9000 as recommended [above](https://github.com/ensenso/ros_driver/issues/118#issuecomment-1833384064) and see if that helps?

belalhmedan90, 2023-12-21:
> roslaunch ensenso_camera \
>      mono_stereo_nodelets.launch \
>      stereo_serial:="232155" \
>      mono_serial:="232155-Color" \
>      node_name_mono:="232155_Color" \ [...]

Please find the [link to that file here](https://drive.google.com/file/d/1y2HVHbAOpTn0nOk-BIlPPI8WXiYPwTk9/view?usp=drive_link) with TCP port set to 25000

belalhmedan90, 2023-12-21:
> I also noticed that your MaximumTransmissionUnit (MTU) is set to 1500. Could you increase it to 9000 as recommended [above](https://github.com/ensenso/ros_driver/issues/118#issuecomment-1833384064) and see if that helps?

I increased it to 9000, but still the error persists.

benthie, 2023-12-21:
> Please find the [link to that file here](https://drive.google.com/file/d/1y2HVHbAOpTn0nOk-BIlPPI8WXiYPwTk9/view?usp=drive_link) with TCP port set to 25000

Unfortunately, the log recorded an open `NxView`, not your ROS node. 

To make things clear, let me explain a little bit. We provide the following software parts:
- `NxLib`: Our library, which can open a TCP port so that one can connect to it
- `NxView`: Our GUI that contains an `NxLib` instance
- `NxTreeEdit`: Our GUI tool that can connect to an `NxLib` instance via TCP and read/write the internal tree/state

What the ROS camera node does is, it starts an `NxLib` instance and opens a camera. And in order to be able to record an `.nxlog` file of what is happening in that `NxLib` , we have to open a TCP port on it. Now we can connect with `NxTreeEdit` to that instance. If you, however, have `NxView` running in parallel, which by default opens port 24000, please make sure to connect to the instance contained within the ROS node (25000 from the example above).

belalhmedan90, 2023-12-21:
> Unfortunately, the log recorded an open `NxView`, not your ROS node.
>
> To make things clear, let me explain a little bit. We provide the following software parts:
>
>     * `NxLib`: Our library, which [...]

sorry for misunderstanding, please find a [correct log file here](https://drive.google.com/file/d/1Xc4Sq9Q-EciN7n6IQ7iYNY-_t2eWeQfi/view?usp=sharing)

benthie, 2023-12-21:
Hi @belalhmedan90 ,

thank you for providing the logs. They helped us identify the problem. I pushed a [patch](https://github.com/ensenso/ros_driver/commit/60b2e79dbc231a8f26bec3614e7e334919b738fe) with which you can set the mono camera capture timeout. Could you pull this version and build it and then try setting a capture timeout for the mono camera by adding the line

```
    <param name="capture_timeout" type="bool" value="1000" />
```

after [line 39](https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/launch/mono_stereo_nodelets.launch#L39) in the `mono_stereo_nodelets.launch` file. This sets the capture timeout for the mono camera to 1000ms. The problem is, that you have FlexView enabled and both stereo and color camera capture about the same point of time but the color image is only delivered after the set of stereo images and thus the mono camera's default (automatic) timeout is not sufficient.

When I tested the driver with a C-Series camera I forgot to also test it with FlexView enabled. You could try to disable FlexView for now to see whether the internal timeout is enough.

Please make sure to rerun `catkin build` after you modified the launch file and pulled the patch. 

We will fix the problem as soon as possible. Again, thank you for your patience.

belalhmedan90, 2023-12-21:
- I pulled the patch, changed the roslaunch file, recompiled my catkin workspace, still the problem not solved, [here is the new log](https://drive.google.com/file/d/1mXdNMRJAJ9X65XOPmgAcQpWoN0hGtKK-/view?usp=sharing), 
> *You could try to disable FlexView*

- Can you please elaborate more, I didn't find that param in the launch file, is there a clear document to explain how to do that? moreover as I understood [from here](https://en.ids-imaging.com/ensenso-3d-camera-operating.html) FlexView is a feature, C57 type camera doesn't support this feature?

benthie, 2023-12-21:
We have a [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/Parameters) on how to read/write parameters from ROS. The quickest way for you right now, however, would be to open NxView, disable FlexView there in the Parameters-Dialog (open either via the Settings-Button or press F8). You might have to switch on the advanced mode. And then save the settings again as JSON and feed it to the ROS node.

I added another line that prints the currently set capture timeout. See [this commit](https://github.com/ensenso/ros_driver/commit/925bb1200338d7f9c6bd2d996f67187861e7a8bf). Please build this version and verify that the output appears in the ROS log of the launch file. In the `.nxlog` file I could see that it takes about 630ms from capture to image arrival, so 1000ms should be fine. You can of course increase the value to 1500 or even 2000 to be safe.

benthie, 2023-12-21:
Hmm, if the nodelet did not print the message, the `ensenso_camera` package was not properly updated or built. How did you set up your catkin workspace? Did you use `git clone`?

You can verify that the package was updated correctly by checking that the file `mono_camera.cpp`on your computer contains the added lines from the patch. 

You can verify that the package builds properly by adding another debug output to e.g. the `color_point_map` script, rebuild your catkin workspace respectively the `ensenso_camera` package and check that added message is printed when you run the script. If not, something is still not properly configured.

belalhmedan90, 2023-12-21:
> Hmm, if the nodelet did not print the message, the `ensenso_camera` package was not properly updated or built. How did you set up your catkin workspace? Did you use `git clone` to install the ensenso `ros_driver`?
> 
> You can verify that the package was updated correctly by checking that the file `mono_camera.cpp`on your computer contains the added lines from the patch.
> 
> You can verify that the package builds properly by adding another debug output to e.g. the `color_point_map` script, rebuild your catkin workspace respectively the `ensenso_camera` package and check that added message is printed when you run the script. If not, something is still not properly configured.

Can you please add this parameter `<param name="capture_timeout" type="int" value="1000" />` to that branch, so I pull, and build without changing many things by hand? thanks

benthie, 2023-12-21:
> Can you please add this parameter `<param name="capture_timeout" type="int" value="1000" />` to the launch file in that branch, so I pull, and build without changing many things by hand? thanks

Sure. See commit c353dd442f4423d7915cb51ace873c24b8e0d4a7.

belalhmedan90, 2023-12-21:
> Sure. See commit [c353dd4](https://github.com/ensenso/ros_driver/commit/c353dd442f4423d7915cb51ace873c24b8e0d4a7).

Thanks, still crashing after few snapshots:
[image]

benthie, 2023-12-21:
Hmm, I am still missing the info message "Timeout set to 1000ms" that should be printed when providing a capture timeout for the mono camera.

Can you do the following inside your workspace:

```
cd src/
git rev-parse HEAD
# Make sure that it is c353dd442f4423d7915cb51ace873c24b8e0d4a7
# Otherwise do
git fetch
git checkout mono-capture-timeout
``

belalhmedan90, 2023-12-21:
> git rev-parse HEAD

I checked the commit, it is the same `c353dd442f4423d7915cb51ace873c24b8e0d4a7`, 

now, if you run the same nodelet on your machine with ubuntu 20.04, cuda acceleration activated (you already have my parameters to reproduce the same output), and C57 camera, does it print that message?

benthie, 2023-12-21:
I will verify that tomorrow. Could you please set the ROS log level to `DEBUG` and provide the verbose logs?

belalhmedan90, 2023-12-21:
> set the ROS log level to `DEBUG`

Please find the [logs here](https://drive.google.com/file/d/16nAcEFIVcVptSeWGeRmCS3fW5bTDIGCJ/view?usp=sharing)

benthie, 2023-12-21:
Please try commit 741de5526ce66623e1ec372e8d69c79ee6d6b88c. This one should now actually use the capture timeout on the mono camera.

belalhmedan90, 2023-12-22:
Indeed, it prints the timeout now, but still crashing after few snapshots:
[image]

benthie, 2023-12-22:
Okay. There is actually a small bug in our Python compatibility layer, hence the `AttributeError` in the script. Fixed in e4b1cc42bf0854b3027757fe9deb37f649980952. But that does not fix the capture timeout.

I currently switched to Ubuntu 22.04 and have no working ROS setup yet. I can try to look into this today, but I do not want to promise that there will be enough time so close to the christmas holidays.

Have you tried reducing the number of images captured by the stereo camera? This can be achieved via the `FlexView` parameter. I just looked into the `ensenso_stereo.json` file that you provided yesterday. There it is set to 4. You can simply turn it off by setting it to `false` as described [here](https://www.ensenso.com/manual/3.5/tree/cameras/120001-%28a-stereo-camera%29/parameters/capture/flexview.html).

Furthermore, please try increasing the capture timeout for the mono camera to 1500 or even 2000 ms until the node runs without capture timeouts.

belalhmedan90, 2023-12-22:
> Okay. There is actually a small bug in our Python compatibility layer, hence the `AttributeError` in the script. Fixed in [e4b1cc4](https://github.com/ensenso/ros_driver/commit/e4b1cc42bf0854b3027757fe9deb37f649980952). But that does not fix the capture timeout.
> 
> I currently switched to Ubuntu 22.04 and have no working ROS setup yet. I can try to look into this today, but I do not want to promise that there will be enough time so close to the christmas holidays.
> 
> Have you tried reducing the number of images captured by the stereo camera? This can be achieved via the `FlexView` parameter. I just looked into the `ensenso_stereo.json` file that you provided yesterday. There it is set to 4. You can simply turn it off by setting it to `false` as described [here](https://www.ensenso.com/manual/3.5/tree/cameras/120001-%28a-stereo-camera%29/parameters/capture/flexview.html).
> 
> Furthermore, please try increasing the capture timeout for the mono camera to 1500 or even 2000 ms until the node runs without capture timeouts.

I tested with FlexView both enabled/disabled, it doesn't solve the problem.

benthie, 2023-12-22:
Okay, that sounds interesting. Could you please upload an `.nxlog` file with `FlexView=false` and timeout 2000?

belalhmedan90, 2023-12-22:
> Okay, that sounds interesting. Could you please upload an `.nxlog` file with `FlexView=false` and timeout 2000?

sorry, my bad, I checked again, the json file that is passed manually to the stereo camera has FlexView enabled, disabled it, the roslaunch file doesn't crash, it works well.
now it turns to be that FlexView is causing the whole issue, as when I disabled it, even 1000ms was enough!

belalhmedan90, 2024-01-19:
This problem is solved with SDK 3.6 and POE++ adapter to power the camera.

belalhmedan90, 2024-01-31:
Hi, this problem appeared with me again while testing the `C57-6-S` Serial number: "233076", I have disabled nxView, increased capture timeout to 2000, here is a full log:
```
started roslaunch server http://NEURA-713:35161/

SUMMARY
========

PARAMETERS
 * /DIP_Register/queue_size: 5
 * /Ensenso_request_data_linked/mono_ns: mono
 * /Ensenso_request_data_linked/stereo_ns: stereo
 * /mono/Ensenso_233076_Color/capture_timeout: 2000
 * /mono/Ensenso_233076_Color/link_frame: 
 * /mono/Ensenso_233076_Color/serial: 233076-Color
 * /mono/Ensenso_233076_Color/settings: 
 * /mono/Ensenso_233076_Color/wait_for_camera: False
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /stereo/Ensenso_233076/camera_frame: 
 * /stereo/Ensenso_233076/link_frame: 
 * /stereo/Ensenso_233076/serial: 233076
 * /stereo/Ensenso_233076/settings: 
 * /stereo/Ensenso_233076/target_frame: 
 * /stereo/Ensenso_233076/tcp_port: -1
 * /stereo/Ensenso_233076/threads: -1
 * /stereo/Ensenso_233076/wait_for_camera: False

NODES
  /
    DIP_Register (nodelet/nodelet)
    Ensenso_request_data_linked (ensenso_camera/request_data_linked)
    manager_ (nodelet/nodelet)
  /mono/
    Ensenso_233076_Color (nodelet/nodelet)
  /stereo/
    Ensenso_233076 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [447366]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to abfb9dd6-c010-11ee-81c7-33c216a155c9
process[rosout-1]: started with pid [447387]
started core service [/rosout]
process[manager_-2]: started with pid [447394]
process[stereo/Ensenso_233076-3]: started with pid [447395]
process[mono/Ensenso_233076_Color-4]: started with pid [447396]
process[Ensenso_request_data_linked-5]: started with pid [447397]
process[DIP_Register-6]: started with pid [447398]
[ INFO] [2024-01-31 08:13:53]: Loading nodelet /stereo/Ensenso_233076 of type ensenso_camera/stereo_camera_node to manager /manager_ with the following remappings:
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2024-01-31 08:13:53]: Loading nodelet /mono/Ensenso_233076_Color of type ensenso_camera/mono_camera_node to manager /manager_ with the following remappings:
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2024-01-31 08:13:53]: Loading nodelet /DIP_Register of type depth_image_proc/register to manager /manager_ with the following remappings:
[ INFO] [2024-01-31 08:13:53]: /depth/camera_info -> /stereo/depth/camera_info
[ INFO] [2024-01-31 08:13:53]: /depth/image_rect -> /stereo/depth/image
[ INFO] [2024-01-31 08:13:53]: /depth_registered/camera_info -> /camera/aligned_depth_to_color/camera_info
[ INFO] [2024-01-31 08:13:53]: /mono/rectified/image -> /mono/depth/image
[ INFO] [2024-01-31 08:13:53]: /rgb/camera_info -> /mono/rectified/camera_info
[ INFO] [2024-01-31 08:13:53]: /stereo/depth/image -> /camera/aligned_depth_to_color/image_raw
[ INFO] [2024-01-31 08:13:53]: Initializing nodelet with 12 worker threads.
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2024-01-31 08:13:53]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2024-01-31 08:13:55]: Opened camera with serial number '233076'.
[ INFO] [2024-01-31 08:13:58]: Opened camera with serial number '233076-Color'.
[ WARN] [2024-01-31 08:13:58]: Camera 233076-Color has an internal link (i.e. it is either extrinsically calibrated (workspace- or hand-eye) or has a link to another camera), but camera and target frame are equal, which means that neither a link nor a target frame has been provided. The images and 3d data retreived from the camera are transformed by the NxLib with the transform stored in the camera's link node, however, this transform is not known to tf. Please provide a link or target frame in order for the transform to be published.
[ERROR] [2024-01-31 08:14:13]: NxLibException 17 (ExecutionFailed) for item /Execute/233076
[ERROR] [2024-01-31 08:14:13]: CaptureTimeout: Waiting for images timed out.
[WARN] [2024-01-31 09:14:13]: Stereo action was not successful.
Traceback (most recent call last):
  File "/home/belalhmedan/ids_ws/devel/lib/ensenso_camera/request_data_linked", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/scripts/request_data_linked", line 65, in <module>
    main()
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/scripts/request_data_linked", line 61, in main
    ros2py.wrap_main_function(_main, "ensenso_camera_request_data_linked")
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/src/ensenso_camera/ros2.py", line 304, in wrap_main_function
    main(node_name)
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/scripts/request_data_linked", line 57, in _main
    ros2py.execute_at_rate(node, main_loop, loop_rate)
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/src/ensenso_camera/ros2.py", line 406, in execute_at_rate
    func()
  File "/home/belalhmedan/ids_ws/src/ensenso_camera/scripts/request_data_linked", line 47, in main_loop
    node.get_logger().error(ros2py.format_error(stereo_result.error, "Error while acquiring stereo data!"))
AttributeError: module 'ensenso_camera.ros2' has no attribute 'format_error'
[Ensenso_request_data_linked-5] process has died [pid 447397, exit code 1, cmd /home/belalhmedan/ids_ws/devel/lib/ensenso_camera/request_data_linked __name:=Ensenso_request_data_linked __log:=/home/belalhmedan/.ros/log/abfb9dd6-c010-11ee-81c7-33c216a155c9/Ensenso_request_data_linked-5.log].
log file: /home/belalhmedan/.ros/log/abfb9dd6-c010-11ee-81c7-33c216a155c9/Ensenso_request_data_linked-5*.log
```

benthie, 2024-01-31:
Are you still on branch [mono-capture-timeout](https://github.com/ensenso/ros_driver/tree/mono-capture-timeout)? Or at least have included commit 647eee0f8c7078864493112953000c04d79a379e in your current branch? I am asking because I am not seeing the capture timeout being printed by the node and I can still see the "AttributeError" bug, which has also been fixed on that branch.

We are soon releasing a bug fix for SDK 3.6 and are planning to release a new version of this driver which includes the mono capture timeout.

belalhmedan90, 2024-01-31:
I'm using the `sdk3.6` branch as I have upgraded my Ensenso SDK to 3.6

benthie, 2024-01-31:
Ah, okay. The `sdk3.6` branch is just a temporary branch which updates the SDK versions tested by the CI. It does not contain the fixes regarding the mono capture timeout.

Please use the `mono-capture-timeout` branch for now. As soon as we have released a new version of this driver including the fixes I will post it here. From there on you can use the `master` branch.

belalhmedan90, 2024-01-31:
This problem appeared again: `roslaunch ensenso_camera color_point_cloud.launch` from my [customized launch file](https://github.com/belalhmedan90/ensenso_ros_driver/blob/mono-capture-timeout/ensenso_camera/launch/color_point_cloud.launch), I'm using the correct branch now `mono-capture-timeout`:
```
SUMMARY
========

PARAMETERS
 * /Ensenso_request_data_linked/mono_ns: mono
 * /Ensenso_request_data_linked/stereo_ns: stereo
 * /color_point_cloud/mono_ns: mono
 * /color_point_cloud/mono_serial: 233076-Color
 * /color_point_cloud/stereo_ns: stereo
 * /mono/Ensenso_233076_Color/capture_timeout: 2000
 * /mono/Ensenso_233076_Color/link_frame: optical_frame_233076
 * /mono/Ensenso_233076_Color/serial: 233076-Color
 * /mono/Ensenso_233076_Color/settings: /home/belalhmedan...
 * /mono/Ensenso_233076_Color/wait_for_camera: False
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /stereo/Ensenso_233076/camera_frame: 
 * /stereo/Ensenso_233076/link_frame: optical_frame_233076
 * /stereo/Ensenso_233076/serial: 233076
 * /stereo/Ensenso_233076/settings: /home/belalhmedan...
 * /stereo/Ensenso_233076/target_frame: 
 * /stereo/Ensenso_233076/tcp_port: 25000
 * /stereo/Ensenso_233076/threads: -1
 * /stereo/Ensenso_233076/wait_for_camera: False

NODES
  /
    Ensenso_request_data_linked (ensenso_camera/request_data_linked)
    color_point_cloud (ensenso_camera/color_point_cloud)
    manager_ (nodelet/nodelet)
  /mono/
    Ensenso_233076_Color (nodelet/nodelet)
  /stereo/
    Ensenso_233076 (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [500891]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a3f77cdc-c028-11ee-81c7-33c216a155c9
process[rosout-1]: started with pid [500913]
started core service [/rosout]
process[manager_-2]: started with pid [500920]
process[stereo/Ensenso_233076-3]: started with pid [500921]
process[mono/Ensenso_233076_Color-4]: started with pid [500922]
process[Ensenso_request_data_linked-5]: started with pid [500923]
process[color_point_cloud-6]: started with pid [500924]
[ INFO] [2024-01-31 11:05:28]: Loading nodelet /stereo/Ensenso_233076 of type ensenso_camera/stereo_camera_node to manager /manager_ with the following remappings:
[ INFO] [2024-01-31 11:05:28]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2024-01-31 11:05:28]: Loading nodelet /mono/Ensenso_233076_Color of type ensenso_camera/mono_camera_node to manager /manager_ with the following remappings:
[ INFO] [2024-01-31 11:05:28]: waitForService: Service [/manager_/load_nodelet] has not been advertised, waiting...
[ INFO] [2024-01-31 11:05:28]: Initializing nodelet with 12 worker threads.
[ INFO] [2024-01-31 11:05:28]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2024-01-31 11:05:28]: waitForService: Service [/manager_/load_nodelet] is now available.
[ INFO] [2024-01-31 11:05:29]: Opened TCP port 25000 on the NxLib.
[ INFO] [2024-01-31 11:05:30]: Opened camera with serial number '233076'.
[ INFO] [2024-01-31 11:05:30]: Loaded settings from '/home/belalhmedan/ids_ws/src/ensenso_camera/config/c_57_6_S_stereo_params.json'.
[ INFO] [2024-01-31 11:05:31]: Opened camera with serial number '233076-Color'.
[ INFO] [2024-01-31 11:05:32]: Loaded settings from '/home/belalhmedan/ids_ws/src/ensenso_camera/config/c_57_6_S_mono_params.json'.
[ INFO] [2024-01-31 11:05:34]: Timeout set to 2000
[ERROR] [2024-01-31 11:05:36]: NxLibException 17 (ExecutionFailed) for item /Execute/233076-Color
[ERROR] [2024-01-31 11:05:36]: CaptureTimeout: Waiting for images timed out.
[ INFO] [2024-01-31 11:05:36]: Timeout set to 2000
[WARN] [2024-01-31 12:05:38]: Mono action was not successful.
[ERROR] [2024-01-31 12:05:38]: Error while acquiring mono data! (Error 17: ExecutionFailed)
[ INFO] [2024-01-31 11:05:38]: Timeout set to 2000
```
please find the [.nxlogs](https://drive.google.com/drive/folders/10uXhC7j2txmd-q-FaoxtpPMUCOg6Tkob?usp=sharing)

benthie, 2024-01-31:
Hmm, I don't see any capture timeouts in the log file. Are you sure you recorded while the node threw the exception?

belalhmedan90, 2024-01-31:
> Hmm, I don't see any capture timeouts in the log file. Are you sure you recorded while the node threw the exception?

Yes, I recorded this nxlogs while the node is running, however I can record a new file.

benthie, 2024-02-06:
The mono camera capture timeout has been merged and is now part of the new [v2.1.2](https://github.com/ensenso/ros_driver/releases/tag/v2.1.2) release.

closed 2024-02-06 by benthie