ensenso/ros_driver #124: Ensenso C-57-6-S Color PointCloud has no color

belalhmedan90, 2024-01-31:
I'm using ROS Noetic, Ubuntu 20.04, with C57-6-S and when launching color_point_cloud, I get a pointcloud without color:
```
roslaunch ensenso_camera color_point_cloud.launch
```
I already hardcoded the params (serial number, json files, etc ...) of my specific camera inside my [customized launch file](https://github.com/belalhmedan90/ensenso_ros_driver/blob/mono-capture-timeout/ensenso_camera/launch/color_point_cloud.launch) in my fork.

[image]

belalhmedan90, 2024-01-31:
Solved by launching the texturing launch file `roslaunch ensenso_camera texture_point_cloud.launch`

benthie, 2024-02-01:
I guess you followed [this tutorial](https://github.com/ensenso/ros_driver/issues/124)?

If so, could you post the output of the two terminals in which you run

1. The camera nodes via 

```
roslaunch ensenso_camera mono_stereo_nodelets.launch \
  stereo_serial:="233076!" \
  mono_serial:="233076-Color" \
  link_frame:="optical_frame_233076"
```
2. The `color_point_cloud` node via 

```
rosrun ensenso_camera color_point_cloud \
  _stereo_serial:="233076!" \
  _mono_serial:="233076-Color"
```

belalhmedan90, 2024-02-01:
> I guess you followed [this tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera)?
> 
> If so, could you post the output of the two terminals in which you run
> 
>     1. The camera nodes via
> 
> 
> ```
> roslaunch ensenso_camera mono_stereo_nodelets.launch \
>   stereo_serial:="233076!" \
>   mono_serial:="233076-Color" \
>   link_frame:="optical_frame_233076"
> ```
> 
>     2. The `color_point_cloud` node via
> 
> 
> ```
> rosrun ensenso_camera color_point_cloud \
>   _stereo_serial:="233076!" \
>   _mono_serial:="233076-Color"
> ```
> 
> Edit: I just saw that the launch file name in the tutorial was wrong. Fixed it.

- The [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera)  was written in a hurry that even the parameters (stereo_serial, mono_serial)are wrong! so I had to investigate the launch files and read the params names.
- Additionally the `color_point_cloud.launch` launches the `mono_stereo_nodelets.launch` and the color_point_cloud node as you can see [here](https://github.com/ensenso/ros_driver/blob/mono-capture-timeout/ensenso_camera/launch/color_point_cloud.launch).
- Can you please do the test on your machine and see if it shows a colored pointcloud on your machine? thanks.

benthie, 2024-02-01:
> * The [tutorial](http://wiki.ros.org/ensenso_driver/Tutorials/C-SeriesCamera)  was written in a hurry that even the parameters (stereo_serial, mono_serial)are wrong! so I had to investigate the launch files and read the params names.

Thank you for the friendly hint. It has been corrected. Next time feel free to open an issue or let us know if something in the tutorials isn't correct, so that errors can be corrected as fast as possible. I tested on both ROS1 and ROS2 and since the parameter names changed from ROS1 to ROS2 I got them mixed up.

> * Additionally the `color_point_cloud.launch` launches the `mono_stereo_nodelets.launch` and the color_point_cloud node as you can see [here](https://github.com/ensenso/ros_driver/blob/mono-capture-timeout/ensenso_camera/launch/color_point_cloud.launch).

Thank you for pointing that out. It is mentioned in the tutorial as an alternative. Maybe something is still wrong in that launch file. I did test the tutorial on ROS1 as I said, but I am not sure whether I also tested this exact launch file.

> * Can you please do the test on your machine and see if it shows a colored pointcloud on your machine?

I just did, although with ROS2, since I switched to a newer Linux where ROS1 is not supported anymore.

If the `color_point_cloud.launch` file does not work could you please run the ` mono_stereo_nodelets.launch` file and the `color_point_cloud` script separately and post the output here. That should help us identify the problem quick.

belalhmedan90, 2024-02-02:
> Hi @belalhmedan90, did you have the time to further investigate the problem?

Hi, since you (the manufacturer) didn't have ROS1 to test the camera ROS driver thoroughly, I gave up testing the camera,  issues still unresolved with ROS1 in my case, and I don't have ROS2.

benthie, 2024-02-02:
Hi [@belalhmedan90](https://github.com/belalhmedan90),

I can understand that it was inconvenient for you that we did not have our ROS driver tested with C-Series cameras when you started testing with the rental camera. But then we tested it and it worked on Ubuntu 20.04 with both ROS1 and ROS2. So yes, we did test on ROS1.

 Right now I am on Ubuntu 22.04 and I only have ROS2, where it still works as expected. Even if I would test it again with ROS1 and it would work, we still wouldn't know why it doesn't work for you.

I believe you that you don't get a colored point cloud, but in order to be able to identify the root cause of the problem, I need to see some logs of the ROS nodes. 

I just had a glance at your [customized launch file](https://github.com/belalhmedan90/ensenso_ros_driver/blob/mono-capture-timeout/ensenso_camera/launch/color_point_cloud.launch) you mentioned above. There you are defining a `mono_capture_timeout`, which - however - is not forwarded to the `mono_stereo_nodelets.launch` file.

Let us know if you resume testing and have any further questions or need help.

Kind Regards
Benny

benthie, 2024-02-06:
I will close this issue for now. Feel free to reopen if the problem occurs again.

closed 2024-02-06 by benthie