ensenso/ros_driver #125: texture_point_cloud The camera node is not running!

belalhmedan90, 2024-01-31:
I'm getting an error: `The camera node is not running!` after launching the texturing launch file, here is the full log, and [nxlogs](https://drive.google.com/file/d/1wmmwnQIkj2NlBDeVaIDEBzF0-B0ZLy_z/view?usp=sharing)
```
SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /texture_point_cloud_image_stream/parameter_set: texture_point_cloud
 * /texture_point_cloud_image_stream/rectified: True

NODES
  /
    texture_point_cloud (ensenso_camera/texture_point_cloud)
    texture_point_cloud_image_stream (ensenso_camera/image_stream)

ROS_MASTER_URI=http://localhost:11311

process[texture_point_cloud-1]: started with pid [533260]
process[texture_point_cloud_image_stream-2]: started with pid [533261]
[ERROR] [2024-01-31 15:24:53]: The camera node is not running!
[texture_point_cloud_image_stream-2] process has finished cleanly
```

benthie, 2024-02-01:
Hi @belalhmedan90,

if you only executed the launch file then this is expected behavior as noted at the top of the launch file: https://github.com/ensenso/ros_driver/blob/master/ensenso_camera/launch/texture_point_cloud.launch#L1

Here's our wiki article on that launch file: http://wiki.ros.org/ensenso_driver/Nodes/texture_point_cloud

What are you trying to achieve, maybe with that info I can help you figure out which nodes you must run and which launch files come in handy for that.

Kind regards
Benny

belalhmedan90, 2024-02-01:
> Hi @belalhmedan90,
>
> if you only executed the launch file then this is expected behavior as noted at the top of the launch file: [...]

I'm running this launch file in addition to the `roslaunch ensenso_camera color_point_cloud.launch`  to get a colored pointcloud.

#124

benthie, 2024-02-01:
The `texture_point_cloud.launch` file runs (a) the `image_stream` script and (b) the `texture_point_cloud` node, which uses the left rectified camera image to add (gray) texture to the point cloud. 

I am a little bit confused: you want to get a colored point cloud and you already ran the `color_point_cloud.launch` file, which gave you a colored point cloud as you wrote yesterday in #124, right? So what is still missing? What is your final goal?

benthie, 2024-02-01:
Okay, then we need to figure out why you are not getting a colored point cloud as expected. I will reopen #124 and close this issue, because it isn't a bug.

closed 2024-02-01 by benthie