ensenso/ros_driver #129: Issues related to integrated development based on X30-5MP [EnsensoSDK version 3.1.996]

caibf, 2024-03-25:
The Capture command in the loop cause such error, our product is installed at the hosptial that may cause electromagnetic interference due to radiation therapy.

I think the cause of the problem is still data transmission problem, but I don't know where the root cause is and how to solve it? cable? Network card? System settings?

[image]
[image]

benthie, 2024-03-25:
Hi @caibf,

I would suggest we shift the discussion to our new [community forum](https://community.ensenso.com/) if that is okay for you. We recently started the forum and want to use it for general issue discussion. This issue tracker then remains for issues related to our ROS interface only.

If it turns out to be ROS related, we can simply switch back to here.

Regards
Benny

caibf, 2024-03-25:
Hello @benthie,

Because the EnsensoSDK is not open source, I am unable to communicate the relevant details with the technical development team in depth, so I can only post again. I am truly sorry.

I am very willing to go to your new forum for communication and discussion, hoping to meet technical experts.

benthie, 2024-03-25:
We - the Ensenso support team - take care of both the community forum and this repository. Information posted to either of the two will be publicly available. 

In case you have confidential information to share, I suggest contacting the [IDS support](https://en.ids-imaging.com/support.html). They will create a ticket and help you with any data / network transmission problems. If it should turn out to be a bug within the Ensenso SDK or the ROS driver, the ticket will be forwarded to the Ensenso support / development team for fixing.