ensenso/ros_driver #142: No ROS 2 nodes for Windows

patrick-muecke-itk, 2025-07-14:
Hi,

could you please provide a convenient way to get ROS 2 nodes for the Ensenso sensors for Windows?

ROS 1 is already past end-of-life and the ROS 2 instructions (https://github.com/ensenso/ros_driver/blob/master/docs/ROS2.md) only work under Linux. 

Please provide documentation and scripts to obtain ROS 2 nodes on Windows (11).

Thank you 🙂

benthie, 2025-07-14:
I just noticed that there is no ROS2 distribution supporting Windows 11 yet, only Windows 10.

patrick-muecke-itk, 2025-07-15:
> I just noticed that there is no ROS2 distribution supporting Windows 11 yet, only Windows 10.
That is correct, but Windows 10 will reach its end-of-life in October, so this will also change very soon probably.
The Ensenso ROS2 build process will hopefully and probably work for both Win 10 and 11.

I'll try to emulate prepare_ros2_build.sh for Windows. Is there a way I can contact you directly in case of problems? Thanks!

patrick-muecke-itk, 2025-07-15:
I have created a batch file `prepare_ros2_build.bat` that works, but I saw that there are actually no requirements for the environment, such that calling `export ROS_VERSION=2 && ./.github/scripts/prepare_ros2_build.sh` in a git bash has the same effect.

However, now I get the following error upon calling `colcon build`:

`
(pixi_ros2_kilted) c:\pixi_ws\ensenso\ros_driver>colcon build
Starting >>> ensenso_camera_msgs
Starting >>> ensenso_description
--- stderr: ensenso_description
CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

---
Failed   <<< ensenso_description [3.61s, exited with code 1]
Aborted  <<< ensenso_camera_msgs [3.69s]

Summary: 0 packages finished [3.89s]
  1 package failed: ensenso_description
  1 package aborted: ensenso_camera_msgs
  2 packages had stderr output: ensenso_camera_msgs ensenso_description
  3 packages not processed
`

I am using ROS2 KK and have MS Visual Studio 2019 installed. cl.exe is in the path. I use the `x64 Native Tools Command Prompt for VS 2019` to start a pixi shell, then source the Ros Environment (call local_setup.bat) and then call `colcon build` from the Ensenso Repo root.

Thanks!

benthie, 2025-07-15:
Have you been able to compile other ROS packages so far in your environment, e.g. one of the examples provided by ROS? A missing compiler to CMake looks like a setup problem to me, because so far no Ensenso package specific errors occurred.

patrick-muecke-itk, 2025-07-15:
Thanks Benny! I indeed get the same error when trying to compile the ROS examples. I'll look into it and notify you how it's going.

patrick-muecke-itk, 2025-07-16:
The CMAKE_CXX_COMPILER  error was due to a somehow broken installation of my Visual Studio. It is fixed now and I got the ROS2 examples to compile completely.

Trying to compile the Ensenso Nodes, ensenso_camera_msgs and ensenso_description compile fine, but for ensenso_camera I get the following error:

> --- stderr: ensenso_camera
> CMake Error at CMakeLists.txt:24 (find_package):
>   By not providing "Findcv_bridge.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "cv_bridge", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "cv_bridge" with
>   any of the following names:
> 
>     cv_bridgeConfig.cmake
>     cv_bridge-config.cmake
> 
>   Add the installation prefix of "cv_bridge" to CMAKE_PREFIX_PATH or set
>   "cv_bridge_DIR" to a directory containing one of the above files.  If
>   "cv_bridge" provides a separate development package or SDK, be sure it has
>   been installed.

benthie, 2025-07-16:
ROS Kilted Kaiju seems to come with OpenCV included, at least the [Jazzy installation page](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html#install-opencv) still had a dedicated on how to install OpenCV and the latest version does not anymore.

The `cv_bridge` is included in the `ensenso_camera` package's manifest (`package.xml`) as a package dependency, so you should be able to install it with [`rosdep`](https://docs.ros.org/en/kilted/Tutorials/Intermediate/Rosdep.html) from within your workspace folder by running: 

~~~
rosdep install --from-paths src -y --ignore-src
~~~

I think you should remove your manually compiled `cv_bridge` first and let `rosdep` do the work for you. At least that's a source of error when it comes to package dependency problems I have encountered in the past.

patrick-muecke-itk, 2025-08-21:
Hi Benny,

I can't find any binaries and I feel like I will not be able to get it working on my end. Could you please get it running on your side and send me detailed instructions?
It would be a great disappointment if there was no option to use Ensenso Sensors on Windows 11 with ROS2.

Thanks!

Patrick

benthie, 2025-08-25:
Hi Patrick,

we decided to not put any effort into supporting ROS2 on Windows for now, because the ecosystem on Windows does not seem to be completely developed yet (missing packages and no `rosdep` to name a few).

While looking for the binaries of the above mentioned packages we stumbled across Robostack. They provide the packages you were looking for
- https://prefix.dev/channels/robostack-kilted/packages/ros-kilted-pcl-conversions
- https://prefix.dev/channels/robostack-kilted/packages/ros-kilted-pcl-ros

but also a complete ROS bundle with a `conda` based package manager. Maybe you can try installing ROS2 via RoboStack and are then able to easily install all missing packages.

Another alternative could be to use WSL.

Kind Regards
Benny